gem5/gem5 #1837: ARM64, mrs %0, cntvct_el0 causes gem5::ArmISA::ISA::getGenericTimer(): Assertion `system' failed

wchen258, 2024-12-04:
**Describe the bug**
```
int cycleLo = 0;  
asm volatile("mrs %0, cntvct_el0" : "=r"(cycleLo));
```
These lines in a test program cause simulator to crash.

**To Reproduce**
```python
import m5
from m5.objects import *

system = System()

system.clk_domain = SrcClockDomain()
system.clk_domain.clock = '1GHz'
system.clk_domain.voltage_domain = VoltageDomain()

system.mem_mode = 'timing'
system.mem_ranges = [AddrRange('512MB')]

system.cpu = ArmTimingSimpleCPU()

system.membus = SystemXBar()

system.cpu.icache_port = system.membus.cpu_side_ports
system.cpu.dcache_port = system.membus.cpu_side_ports

system.cpu.createInterruptController()

system.system_port = system.membus.cpu_side_ports

system.mem_ctrl = MemCtrl()
system.mem_ctrl.dram = DDR3_1600_8x8()
system.mem_ctrl.dram.range = system.mem_ranges[0]
system.mem_ctrl.port = system.membus.mem_side_ports

binary = '<path to test program>'

# for gem5 V21 and beyond
system.workload = SEWorkload.init_compatible(binary)

process = Process()
process.cmd = [binary]
system.cpu.workload = process
system.cpu.createThreads()

root = Root(full_system = False, system = system)
m5.instantiate()

print("Beginning simulation!")
exit_event = m5.simulate()

print('Exiting @ tick {} because {}'.format(m5.curTick(), exit_event.getCause()))
```

```test-program.c
#include <stdio.h>

int main(int argc, char* argv[])
{
    int cycleLo = 0;  
    asm volatile("mrs %0, cntvct_el0" : "=r"(cycleLo));
    printf("LETS GOOO!!!\n");
    return 0;
}
```

**Terminal Output**
```
$ build/ARM/gem5.opt configs/tutorial/part1/simple.py
gem5 Simulator System.  https://www.gem5.org
gem5 is copyrighted software; use the --copyright option for details.

gem5 version 24.0.0.1
gem5 compiled Dec  4 2024 15:59:58
gem5 started Dec  4 2024 18:41:36
gem5 executing on BillyGates, pid 1478
command line: build/ARM/gem5.opt configs/tutorial/part1/simple.py

Global frequency set at 1000000000000 ticks per second
warn: No dot file generated. Please install pydot to generate the dot file and pdf.
src/mem/dram_interface.cc:690: warn: DRAM device capacity (8192 Mbytes) does not match the address range assigned (512 Mbytes)
src/base/statistics.hh:279: warn: One of the stats is a legacy stat. Legacy stat is a stat that does not belong to any statistics::Group. Legacy stat is deprecated.
system.remote_gdb: Listening for connections on port 7000
Beginning simulation!
src/sim/simulate.cc:199: info: Entering event queue @ 0.  Starting simulation...
src/sim/syscall_emul.cc:74: warn: ignoring syscall set_robust_list(...)
src/sim/syscall_emul.cc:85: warn: ignoring syscall rseq(...)
      (further warnings will be suppressed)
src/sim/mem_state.cc:448: info: Increasing stack size by one page.
src/sim/syscall_emul.hh:1075: warn: readlink() called on '/proc/self/exe' may yield unexpected results in various settings.
      Returning '/mnt/c/Users/User/Desktop/gem5_test/gem5/tests/test-progs/hello/src/hello64-static'
src/arch/arm/insts/pseudo.cc:174: warn:         instruction 'bti' unimplemented
src/sim/syscall_emul.cc:74: warn: ignoring syscall mprotect(...)
gem5.opt: src/arch/arm/isa.cc:1396: gem5::ArmISA::BaseISADevice& gem5::ArmISA::ISA::getGenericTimer(): Assertion `system' failed.
Program aborted at tick 315627000
--- BEGIN LIBC BACKTRACE ---
build/ARM/gem5.opt(_ZN4gem515print_backtraceEv+0x30)[0x562e38979450]
build/ARM/gem5.opt(_ZN4gem512abortHandlerEi+0x4c)[0x562e3899d9ac]
/lib/x86_64-linux-gnu/libc.so.6(+0x42520)[0x7ff8c5289520]
/lib/x86_64-linux-gnu/libc.so.6(pthread_kill+0x12c)[0x7ff8c52dd9fc]
/lib/x86_64-linux-gnu/libc.so.6(raise+0x16)[0x7ff8c5289476]
/lib/x86_64-linux-gnu/libc.so.6(abort+0xd3)[0x7ff8c526f7f3]
/lib/x86_64-linux-gnu/libc.so.6(+0x2871b)[0x7ff8c526f71b]
/lib/x86_64-linux-gnu/libc.so.6(+0x39e96)[0x7ff8c5280e96]
build/ARM/gem5.opt(_ZN4gem56ArmISA3ISA15getGenericTimerEv+0x19f)[0x562e369c43af]
build/ARM/gem5.opt(_ZN4gem56ArmISA3ISA11readMiscRegEt+0x301)[0x562e369c92b1]
build/ARM/gem5.opt(_ZNK4gem510ArmISAInst5Mrs647executeEPNS_11ExecContextEPNS_5trace10InstRecordE+0x55)[0x562e372ed485]
build/ARM/gem5.opt(_ZN4gem515TimingSimpleCPU14completeIfetchEPNS_6PacketE+0x264)[0x562e37c83b94]
build/ARM/gem5.opt(_ZN4gem510EventQueue10serviceOneEv+0x16d)[0x562e3898dd5d]
build/ARM/gem5.opt(_ZN4gem59doSimLoopEPNS_10EventQueueE+0x68)[0x562e389b4c38]
build/ARM/gem5.opt(_ZN4gem58simulateEm+0x283)[0x562e389b5233]
build/ARM/gem5.opt(+0x30d8510)[0x562e38906510]
build/ARM/gem5.opt(+0x132fdc4)[0x562e36b5ddc4]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x128023)[0x7ff8c640f023]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyObject_Call+0x5c)[0x7ff8c63c8fec]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x4b16)[0x7ff8c635d776]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af)[0x7ff8c64a93af]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9d68)[0x7ff8c63629c8]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af)[0x7ff8c64a93af]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(PyEval_EvalCode+0xbe)[0x7ff8c64a43de]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1bd96d)[0x7ff8c64a496d]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1287b3)[0x7ff8c640f7b3]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x69de)[0x7ff8c635f63e]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af)[0x7ff8c64a93af]
build/ARM/gem5.opt(+0x30be9d7)[0x562e388ec9d7]
build/ARM/gem5.opt(main+0x213)[0x562e3690c8e3]
/lib/x86_64-linux-gnu/libc.so.6(+0x29d90)[0x7ff8c5270d90]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80)[0x7ff8c5270e40]
--- END LIBC BACKTRACE ---
For more info on how to address this issue, please visit https://www.gem5.org/documentation/general_docs/common-errors/

Aborted (core dumped)
```

**Host Operating System**
WSL 2 (Ubuntu 22.04.4 LTS)

**Host ISA**
X86

Any help is appreciated, thank you in advance!

giactra, 2024-12-05:
We should probably make the error more informative.

However, the problem is that there is no GenericTimer in SE mode. Do you have an userspace process trying to read the virtual count register? I guess if that is the case we could mock something up to give some idea of time to the application

wchen258, 2024-12-05:
Thank you for quick reply. Yea, we have a benchmark that tries to read the timer and we cannot really change the source. If you can tell us how we can get it to work with some mock timer data, it would be great!

giactra, 2024-12-05:
You probably need to define a SE generic timer inheriting from BaseISADevice.
Then you would return a reference to it from the ISA::getGenericTimer method.

You only need to implement reads to EL0 registers. In your case, I would for example use the curTick time from the CPU clock domain, and transform tick time to a counter value depending on the counting frequency. (You can copy the default value of operation of the SystemCounter: https://github.com/gem5/gem5/blob/stable/src/dev/arm/GenericTimer.py#L70 )

Feel free to open a PR one you implement this :)

wchen258, 2024-12-06:
Thank you very much for the info! This issue does not directly relate to our current project, but will take a look into it once we are more familiar with Gem5 (we are true beginners), and will open a PR if we feel competent enough.

closed 2024-12-06 by wchen258

giactra, 2024-12-06:
> Thank you very much for the info! This issue does not directly relate to our current project, but will take a look into it once we are more familiar with Gem5 (we are true beginners), and will open a PR if we feel competent enough.

Sure! Please don't feel intimidated by opening a PR. We have all been beginners at a certain point in time. I'd be happy to review it and provide assistance if needed