gem5/gem5 #1856: [RISC-V] Incorrect results on latest stable branch (commit 63d25922)

psharath1, 2024-12-13:
**Describe the bug**
The latest Stable branch (commit 63d25922) is giving incorrect results

**Affects version**
stable branch commit 63d25922

**gem5 Modifications**
None

**To Reproduce**
Steps to reproduce the behavior. Please assume starting from a clean repository:

1. Compile gem5 with command ...  scons build/RISCV/gem5.opt

```
uint16_t sumOfElements(const uint8_t* src,size_t nLength) 
{
  size_t processed_size = 0;
  size_t output_size=(nLength);
  size_t output_loop_size=(output_size/32);
  uint8_t* in_ptr = (uint8_t *)src;

  vuint8m1_t  vecIn;
  vuint16m2_t vecAcc;
  
  size_t vlmax = __riscv_vsetvl_e8m1(nLength);
  vecAcc = __riscv_vmv_v_x_u16m2(0, vlmax);
  vuint16m1_t redSum =  __riscv_vmv_v_x_u16m1(0, vlmax);
  size_t vl;

  do
  {
	  vl = __riscv_vsetvl_e8m1(nLength - processed_size);
	  vecIn = __riscv_vle8_v_u8m1(in_ptr, vl);
	  vecAcc = __riscv_vwaddu_wv_u16m2(vecAcc, vecIn, vl);
	  in_ptr += vl;
	  processed_size += vl;
  }while(processed_size<nLength);
  redSum =  __riscv_vredsum_vs_u16m2_u16m1(vecAcc, redSum, vlmax);
  uint16_t sum = __riscv_vmv_x_s_u16m1_u16(redSum);
  return sum;
}
```
**Expected behavior**
Spike and gem5 branch commit b1a44b89c7bae73fae2dc547bc1f871452075b85 (HEAD -> stable, tag: v24.0.0.1, tag: v24.0, origin/stable, origin/HEAD) gives expected results

Trace from gem5 tag: v24.0.0.1
```
6295000: system.cpu: T0 : 0x10304 @_Z13sumOfElementsPKhm    : vsetvli a3, a1, e16, m2, ta, ma : SimdConfig :  D=0x0000000000000020
6353750: system.cpu: T0 : 0x10308 @_Z13sumOfElementsPKhm+4    : vle8_v v1, (a0)           
6353750: system.cpu: T0 : 0x10308 @_Z13sumOfElementsPKhm+4. 0 : vle8_v_micro v1, 0(a0), v1 : SimdUnitStrideLoad :  D=[00010203_04050607_08090a0b_0c0d0e0f_10111213_14151617_18191a1b_1c1d1e1f] A=0x1c160
6360000: system.cpu: T0 : 0x1030c @_Z13sumOfElementsPKhm+8    : vmv_v_i v6, v6, 0         
6360000: system.cpu: T0 : 0x1030c @_Z13sumOfElementsPKhm+8. 0 : vmv_v_i_micro v6, v0, 0    : SimdMisc :  D=[00000000_00000000_00000000_00000000_00000000_00000000_00000000_00000000]
6361250: system.cpu: T0 : 0x1030c @_Z13sumOfElementsPKhm+8. 1 : vmv_v_i_micro v7, v1, 0    : SimdMisc :  D=[00000000_00000000_00000000_00000000_00000000_00000000_00000000_00000000]
6362500: system.cpu: T0 : 0x10310 @_Z13sumOfElementsPKhm+12    : vsetvli zero, a3, e16, m1, ta, ma : SimdConfig :  D=0x0000000000000010
6432500: system.cpu: T0 : 0x10314 @_Z13sumOfElementsPKhm+16    : vmv_v_i v4, v4, 0         
6432500: system.cpu: T0 : 0x10314 @_Z13sumOfElementsPKhm+16. 0 : vmv_v_i_micro v4, v0, 0    : SimdMisc :  D=[00000000_00000000_00000000_00000000_00000000_00000000_00000000_00000000]
6433750: system.cpu: T0 : 0x10318 @_Z13sumOfElementsPKhm+20    : vsetvli zero, a3, e8, m1, ta, ma : SimdConfig :  D=0x0000000000000020
6445000: system.cpu: T0 : 0x1031c @_Z13sumOfElementsPKhm+24    : c_add a0, a3               : IntAlu :  D=0x000000000001c180
6448750: system.cpu: T0 : 0x1031e @_Z13sumOfElementsPKhm+26    : vwaddu_wv v2, v6, v1      
6448750: system.cpu: T0 : 0x1031e @_Z13sumOfElementsPKhm+26. 0 : vwaddu_wv_micro v2, v6, v1 : SimdAdd :  D=[00000100_02000300_04000500_06000700_08000900_0a000b00_0c000d00_0e000f00]
6450000: system.cpu: T0 : 0x1031e @_Z13sumOfElementsPKhm+26. 1 : vwaddu_wv_micro v3, v7, v1 : SimdAdd :  D=[10001100_12001300_14001500_16001700_18001900_1a001b00_1c001d00_1e001f00]
6450000: system.cpu: T0 : 0x10322 @_Z13sumOfElementsPKhm+30    : bgeu a3, a1, 34            : IntAlu : 
6463750: system.cpu: T0 : 0x10344 @_Z13sumOfElementsPKhm+64    : vsetvli zero, a3, e16, m2, ta, ma : SimdConfig :  D=0x0000000000000020
6477500: system.cpu: T0 : 0x10348 @_Z13sumOfElementsPKhm+68    : vredsum_vs v2, v2, v4     
6477500: system.cpu: T0 : 0x10348 @_Z13sumOfElementsPKhm+68. 0 : vredsum_vs_micro v2, v2, v4 : SimdReduceAdd :  D=[78000100_02000300_04000500_06000700_08000900_0a000b00_0c000d00_0e000f00]
6482500: system.cpu: T0 : 0x10348 @_Z13sumOfElementsPKhm+68. 1 : vredsum_vs_micro v2, v3, v4 : SimdReduceAdd :  D=[f0010100_02000300_04000500_06000700_08000900_0a000b00_0c000d00_0e000f00]
6487500: system.cpu: T0 : 0x1034c @_Z13sumOfElementsPKhm+72    : vmv_x_s a0, v2             : SimdMisc :  D=0x00000000000001f0
6488750: system.cpu: T0 : 0x10350 @_Z13sumOfElementsPKhm+76    : c_slli a0, 48              : IntAlu :  D=0x01f0000000000000
6490000: system.cpu: T0 : 0x10352 @_Z13sumOfElementsPKhm+78    : c_srli a0, 48              : IntAlu :  D=0x00000000000001f0
6490000: system.cpu: T0 : 0x10354 @_Z13sumOfElementsPKhm+80    : c_jr ra                    : IntAlu : 
```
Trace from gem5 tag: tag: v24.1.0.0
```
6295000: system.cpu: T0 : 0x10304 @_Z13sumOfElementsPKhm    : vsetvli a3, a1, e16, m2, ta, ma : SimdConfig :  D=0x0000000000000020
6353750: system.cpu: T0 : 0x10308 @_Z13sumOfElementsPKhm+4    : vle8_v v1, (a0)           
6353750: system.cpu: T0 : 0x10308 @_Z13sumOfElementsPKhm+4. 0 : vle8_v_micro v1, 0(a0), zero : SimdUnitStrideLoad :  D=[00010203_04050607_08090a0b_0c0d0e0f_10111213_14151617_18191a1b_1c1d1e1f] A=0x1c160
6360000: system.cpu: T0 : 0x1030c @_Z13sumOfElementsPKhm+8    : vmv_v_i v6, v6, 0         
6360000: system.cpu: T0 : 0x1030c @_Z13sumOfElementsPKhm+8. 0 : vmv_v_i_micro v6, v0, 0    : SimdMisc :  D=[00000000_00000000_00000000_00000000_00000000_00000000_00000000_00000000]
6361250: system.cpu: T0 : 0x1030c @_Z13sumOfElementsPKhm+8. 1 : vmv_v_i_micro v7, v1, 0    : SimdMisc :  D=[00000000_00000000_00000000_00000000_00000000_00000000_00000000_00000000]
6362500: system.cpu: T0 : 0x10310 @_Z13sumOfElementsPKhm+12    : vsetvli zero, a3, e16, m1, ta, ma : SimdConfig :  D=0x0000000000000010
6432500: system.cpu: T0 : 0x10314 @_Z13sumOfElementsPKhm+16    : vmv_v_i v4, v4, 0         
6432500: system.cpu: T0 : 0x10314 @_Z13sumOfElementsPKhm+16. 0 : vmv_v_i_micro v4, v0, 0    : SimdMisc :  D=[00000000_00000000_00000000_00000000_00000000_00000000_00000000_00000000]
6433750: system.cpu: T0 : 0x10318 @_Z13sumOfElementsPKhm+20    : vsetvli zero, a3, e8, m1, ta, ma : SimdConfig :  D=0x0000000000000020
6445000: system.cpu: T0 : 0x1031c @_Z13sumOfElementsPKhm+24    : c_add a0, a3               : IntAlu :  D=0x000000000001c180
6448750: system.cpu: T0 : 0x1031e @_Z13sumOfElementsPKhm+26    : vwaddu_wv v2, v6, v1      
6448750: system.cpu: T0 : 0x1031e @_Z13sumOfElementsPKhm+26. 0 : vwaddu_wv_micro v2, v6, v1 : SimdAdd :  D=[00000100_02000300_04000500_06000700_08000900_0a000b00_0c000d00_0e000f00]
6450000: system.cpu: T0 : 0x1031e @_Z13sumOfElementsPKhm+26. 1 : vwaddu_wv_micro v3, v7, v1 : SimdAdd :  D=[10001100_12001300_14001500_16001700_18001900_1a001b00_1c001d00_1e001f00]
6450000: system.cpu: T0 : 0x10322 @_Z13sumOfElementsPKhm+30    : bgeu a3, a1, 34            : IntAlu : 
6463750: system.cpu: T0 : 0x10344 @_Z13sumOfElementsPKhm+64    : vsetvli zero, a3, e16, m2, ta, ma : SimdConfig :  D=0x0000000000000020
6477500: system.cpu: T0 : 0x10348 @_Z13sumOfElementsPKhm+68    : vredsum_vs v2, v2, v4     
6477500: system.cpu: T0 : 0x10348 @_Z13sumOfElementsPKhm+68. 0 : vredsum_vs_micro v2, v2, v4 : SimdReduceAdd :  D=[7800ffff_ffffffff_ffffffff_ffffffff_ffffffff_ffffffff_ffffffff_ffffffff]
6478750: system.cpu: T0 : 0x10348 @_Z13sumOfElementsPKhm+68. 1 : vredsum_vs_micro v2, v3, v4 : SimdReduceAdd :  D=[7801ffff_ffffffff_ffffffff_ffffffff_ffffffff_ffffffff_ffffffff_ffffffff]
6483750: system.cpu: T0 : 0x1034c @_Z13sumOfElementsPKhm+72    : vmv_x_s a0, v2             : SimdMisc :  D=0x0000000000000178
6485000: system.cpu: T0 : 0x10350 @_Z13sumOfElementsPKhm+76    : c_slli a0, 48              : IntAlu :  D=0x0178000000000000
6486250: system.cpu: T0 : 0x10352 @_Z13sumOfElementsPKhm+78    : c_srli a0, 48              : IntAlu :  D=0x0000000000000178
6486250: system.cpu: T0 : 0x10354 @_Z13sumOfElementsPKhm+80    : c_jr ra                    : IntAlu : 
```
Issue is in 
```
6477500: system.cpu: T0 : 0x10348 @_Z13sumOfElementsPKhm+68    : vredsum_vs v2, v2, v4     
6477500: system.cpu: T0 : 0x10348 @_Z13sumOfElementsPKhm+68. 0 : vredsum_vs_micro v2, v2, v4 : SimdReduceAdd :  D=[7800ffff_ffffffff_ffffffff_ffffffff_ffffffff_ffffffff_ffffffff_ffffffff]
6478750: system.cpu: T0 : 0x10348 @_Z13sumOfElementsPKhm+68. 1 : vredsum_vs_micro v2, v3, v4 : SimdReduceAdd :  D=[7801ffff_ffffffff_ffffffff_ffffffff_ffffffff_ffffffff_ffffffff_ffffffff]
```
vredsum_vs_micro v2, v3, v4 is not adding the reduced sum from v3 and v4 to v2 as required.
**Host Operating System**
Ubuntu 22.04

**Host ISA**
RISC-V

**Compiler used**
riscv64-unknown-elf-gcc

**Additional information**
 We are also noticing this behaviour with amny other instructions as well.

Harshil2107, 2024-12-17:
Hi, thank you for highlighting this issue, 

I am trying to reproduce this error, I had a few questions:

- I assume the C code in the issue is the binary that is run in the simulation, Can you give me the entire script or the steps to get the final binary like the function arguments, imports and the build command

- Can you also give me the gem5 configuration script that you used so that I can re run with the same configuration.

psharath1, 2024-12-17:
[Test.zip](https://github.com/user-attachments/files/18159503/Test.zip)

The attached zip file contains the necessary files.

The commands used as are as given below
riscv64-unknown-elf-gcc -march=rv64imafdcv -mabi=lp64d -O2 -ftree-vectorize -fno-tree-loop-if-convert benchmark.cc  -o TestWideningAdd
Spike Simulation
spike --isa=rv64imafdcv_zvl256b_zve32d_zicntr pk64 TestWideningAdd
gem5 simulation (Please check the paths)
build/RISCV/gem5.opt simple-Minor.py --binary=TestWideningAdd

have also included a makefile for building the binary

Thanks,
Sharath

powerjg, 2025-01-31:
Is this the same issue? https://github.com/gem5/gem5/pull/1955#issuecomment-2627471711

psharath1, 2025-02-01:
> Is this the same issue? [#1955 (comment)](https://github.com/gem5/gem5/pull/1955#issuecomment-2627471711)

yes it is :)

erin-le, 2025-06-12:
@psharath1: Hello, since all of the PRs related to this issue have been merged, could you check if there are still issues on your end? Thank you!

psharath1, 2025-06-13:
@erin-le I can confirm that the issues that I was seeing before have all been fixed on the latest branch.
Thanks,
Sharath