gem5/gem5 #1888: RISCV Dynamic link for FS mode

liuyh-beep, 2024-12-27:
Hi,

I am a beginner to learn gem5. I want to know how to run a dynamically linked ELF file under FS mode? And some details:

```bash
root@f9b40f333ee9:/gem5_work/gem5# file ./rvv_test/bin/matmul_kernel_dy.elf 
./rvv_test/bin/matmul_kernel_dy.elf: ELF 64-bit LSB pie executable, UCB RISC-V, RVC, double-float ABI, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-riscv64-lp64d.so.1, for GNU/Linux 4.15.0, with debug_info, not stripped
```

Can anyone provide some materials about dynamic link setting?

Thank you in advance.

powerjg, 2024-12-27:
If you're running gem5 in full system mode and the compiler toolchain's dynamic libraries are installed on the disk, this should work. If you're using busybox you probably won't be able to use dynamically linked executables. However, if you use a full ubuntu distribution, you should have no problems. See our ubuntu resources for RISC-V here: https://resources.gem5.org/resources?q=architecture%3ARISCV+ubuntu&page=1&sort=relevance&limit=10

liuyh-beep, 2024-12-28:
Thank you for your sharing!🤗