gem5/gem5 #1903: Potential memory leaks

powerjg, 2025-01-06:
In trying to track down the memory usage increase on x86 boot, I ran valgrind. I didn't find the culprit to our memory usage increase, but I did find some potential memory leaks. Noting them here.

I've also attached the full valgrind output.

### This one shows up a couple of times. Seems like a real leak.

Shows up both on the read and the write path

```
==1267085== 72 bytes in 1 blocks are possibly lost in loss record 16,205 of 29,787
==1267085==    at 0x72BFFA3: operator new(unsigned long) (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==1267085==    by 0x34760D0: gem5::TimingSimpleCPU::handleReadPacket(gem5::Packet*) (timing.cc:282)
```

```
==1267085== 42,192 bytes in 586 blocks are definitely lost in loss record 29,253 of 29,787
==1267085==    at 0x72BFFA3: operator new(unsigned long) (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==1267085==    by 0x3476975: gem5::TimingSimpleCPU::handleWritePacket() (timing.cc:511)
==1267085==    by 0x347ACB5: gem5::TimingSimpleCPU::sendData(std::shared_ptr<gem5::Request> const&, unsigned char*, unsigned long*, bool) (timing.cc:336)
```

https://github.com/gem5/gem5/blob/develop/src/cpu/simple/timing.cc#L282
https://github.com/gem5/gem5/blob/develop/src/cpu/simple/timing.cc#L511

Note: There are multiple paths to get to these `new` calls.

### These are likely missing a delete at the same place

```
==1267085== 472 (176 direct, 296 indirect) bytes in 1 blocks are definitely lost in loss record 25,224 of 29,787
==1267085==    at 0x72BFFA3: operator new(unsigned long) (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==1267085==    by 0x403DE31: gem5::X86ISA::Walker::WalkerState::setupWalk(unsigned long) (pagetable_walker.cc:607)
==1267085==    by 0x404030F: gem5::X86ISA::Walker::WalkerState::startWalk() (pagetable_walker.cc:235)
```

```
==1267085== 472 (176 direct, 296 indirect) bytes in 1 blocks are definitely lost in loss record 25,225 of 29,787
==1267085==    at 0x72BFFA3: operator new(unsigned long) (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==1267085==    by 0x403EBE1: gem5::X86ISA::Walker::WalkerState::stepWalk(gem5::Packet*&) (pagetable_walker.cc:531)
==1267085==    by 0x404128A: gem5::X86ISA::Walker::WalkerState::recvPacket(gem5::Packet*) (pagetable_walker.cc:634)
```

https://github.com/gem5/gem5/blob/develop/src/arch/x86/pagetable_walker.cc#L607
https://github.com/gem5/gem5/blob/develop/src/arch/x86/pagetable_walker.cc#L531

### In the IDE controller

This *could* be our culprit since we leaked 128KiB in one leak. Maybe when we're reading more off of the disk during systemd boot this is leaking more? It only happens on resetting the device state, but that happens in `updateState` on line 780. I wonder if something changed so that resetting is happening more often now?

```
==1267085== 131,072 bytes in 1 blocks are possibly lost in loss record 29,548 of 29,787
==1267085==    at 0x72C15C3: operator new[](unsigned long) (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==1267085==    by 0x29D7D89: gem5::IdeDisk::reset(int) (ide_disk.cc:148)
==1267085==    by 0x29DC02C: gem5::IdeDisk::IdeDisk(gem5::IdeDiskParams const&) (ide_disk.cc:76)
```

https://github.com/gem5/gem5/blob/develop/src/dev/storage/ide_disk.cc#L148

### These are just done once at initialization or a small number of times during simulation

```
==1267085== 80 bytes in 2 blocks are definitely lost in loss record 16,893 of 29,787
==1267085==    at 0x72BFFA3: operator new(unsigned long) (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==1267085==    by 0x4012B66: gem5::X86ISA::ISA::newPCState(unsigned long) const (isa.hh:69)
==1267085==    by 0x329EFCC: clearArchRegs (simple_thread.hh:247)
```

```
==1267085== 360 (64 direct, 296 indirect) bytes in 1 blocks are definitely lost in loss record 24,739 of 29,787
==1267085==    at 0x72BFFA3: operator new(unsigned long) (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==1267085==    by 0x315E2A9: BaseGlobalEventTemplate (global_event.hh:163)
==1267085==    by 0x315E2A9: GlobalEvent (global_event.hh:197)
==1267085==    by 0x315E2A9: gem5::GlobalSimLoopExitEvent::GlobalSimLoopExitEvent(unsigned long, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, unsigned long) (sim_events.cc:59)
==1267085==    by 0x315E9D7: gem5::exitSimLoop(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, unsigned long, unsigned long, bool) (sim_events.cc:95)
==1267085==    by 0x31CB065: gem5::pseudo_inst::m5exit(gem5::ThreadContext*, unsigned long) (pseudo_inst.cc:183)
```

[valgrind.txt](https://github.com/user-attachments/files/18322326/valgrind.txt)

ncppd, 2025-02-25:
I discovered that in RISC-V full system simulations, a similar issue occurs with the O3 CPU models (and possibly with MINOR as well). I set up a spec2017-test simulation, and after about one day, the physical memory usage of 605.mcf had grown to 50GB - far exceeding the 16GB allocated for the simulation. ATOMIC/TIMING models seem fine and never exceed the 16G mark.

powerjg, 2025-02-25:
That's good to know. I found many other memory "issues" (most actually turned out to not be *leaks* per se) by running massif. 

I doubt that I will have the bandwidth to dig deeper into this issue any time soon, but if you want to look into it, I suggest using massif. 

See #1957 and #1945 for two examples.