gem5/gem5 #1941: [bug] [arm]： pfr0_el1.gic  will never be true because of the value is used before it can be initiated

gxflying, 2025-01-21:
the  value is initialized in [preUnflattenMiscReg()](https://github.com/gem5/gem5/blob/c9625ce9cc5b5a90a38327de5ac0e1870974af5e/src/arch/arm/isa.cc#L132C4-L132C26l)  in class ISA  constructor
[pfr0_el1.gic = FullSystem && getGICv3CPUInterface(tc) ? 0x1 : 0;](https://github.com/gem5/gem5/blob/c9625ce9cc5b5a90a38327de5ac0e1870974af5e/src/arch/arm/regs/misc.cc#L4906)

the question is when ISA is created and constructor is called  the [ISA::getGICv3CPUInterface(ThreadContext *tc)](https://github.com/gem5/gem5/blob/c9625ce9cc5b5a90a38327de5ac0e1870974af5e/src/arch/arm/isa.cc#L1385) **will always return nullptr**, because at this time the gic has NOT been created yet !!

```
ISA::getGICv3CPUInterface(ThreadContext *tc)
{
    assert(system);
    Gicv3 *gicv3 = dynamic_cast<Gicv3 *>(system->getGIC());    //  gic has NOT been created when ISA constructor is called
    if (gicv3) {
        return gicv3->getCPUInterface(tc->contextId());
    } else {
        return nullptr;
    }
}
```
one solution for this might be put the [preUnflattenMiscReg()](https://github.com/gem5/gem5/blob/c9625ce9cc5b5a90a38327de5ac0e1870974af5e/src/arch/arm/isa.cc#L132C4-L132C26l) in ISA::init or ISA::initState which will be called after all the simobject is created

### Code snippet from [simulate.py :: instantiate](https://github.com/gem5/gem5/blob/c9625ce9cc5b5a90a38327de5ac0e1870974af5e/src/python/m5/simulate.py#L80)

```
    # Create the C++ sim objects and connect ports
    for obj in root.descendants():
        obj.createCCObject()
    for obj in root.descendants():
        obj.connectPorts()

    # Do a second pass to finish initializing the sim objects
    for obj in root.descendants():
        obj.init()                                                  //   call preUnflattenMiscReg()  to init arm register here after  all simobjects created !
....
    # Restore checkpoint (if any)
    if ckpt_dir:
        _drain_manager.preCheckpointRestore()
        ckpt = _m5.core.getCheckpoint(ckpt_dir)
        for obj in root.descendants():
            obj.loadState(ckpt)
    else:
        for obj in root.descendants():
            obj.initState()
```

erin-le, 2025-01-23:
@gxflying Hello, thank you for submitting this bug report. Since you have insight into the problem and a potential solution, would you be willing to work on a fix and submit a PR for this issue?

gxflying, 2025-01-26:
> [@gxflying](https://github.com/gxflying) Hello, thank you for submitting this bug report. Since you have insight into the problem and a potential solution, would you be willing to work on a fix and submit a PR for this issue?

I have test the solution for my use case， should I commit the change directly ？

erin-le, 2025-01-30:
Hello @gxflying, please open a PR with your solution and target the `develop` branch. In the PR title, please include one or more of the tags here: https://github.com/gem5/gem5/labels, followed by a colon and the title of the PR. In this case, `arch-arm` would be a good label to use. Please also follow this format for the titles of your commit messages. 

For the commit messages, please have a title, then leave a blank line, then go into additional detail about the changes in the commit. This guide on contributing to gem5: https://bootcamp.gem5.org/#06-Contributing/01-contributing has instructions on how to format the commit message on pages 32 and 33. There are also instructions for installing pre-commit, which helps with Python formatting, as well as some C++ style guidelines. Note that this guide is slightly out of date, and that we no longer require Change-Ids.

For more examples on formatting, you can also look at PRs and commits in the repo.

Thanks for your contribution!

gxflying, 2025-02-07:
I don't know how to add labels for a PR -_-!  , and the PR is without a label  : https://github.com/gem5/gem5/pull/1980