gem5/gem5 #1951: arch-x86: `PACKUSWB` produces incorrect results

nmosier, 2025-01-28:
**Describe the bug**
gem5' implementation of the `PACKUSWB` instruction does not produce the correct output (when compared to what real x86 hardware produces).

Here is a proof-of-concept program to reproduce the bug:
```c
#include <stdio.h>
#include <stdint.h>

static uint64_t test(uint64_t a, uint64_t b) {
  uint64_t c;
  asm volatile ("movq %1, %%mm0\n"
                "movq %2, %%mm1\n"
                "packuswb %%mm1, %%mm0\n"
                "movq %%mm0, %0\n"
                : "=r"(c)
                : "r"(a), "r"(b));
  return c;
}

static void do_test(uint64_t a, uint64_t b) {
  const uint64_t c = test(a, b);
  printf("%016lx\n"
         "%016lx\n"
         "%016lx\n",
         a, b, c);
}

int main() {
  do_test(0x80, 0);
  do_test(0xF000, 0);
  do_test(0xFFFF, 0);
}
```

On the host, it prints:
```
0000000000000080
0000000000000000
0000000000000080
000000000000f000
0000000000000000
0000000000000000
000000000000ffff
0000000000000000
0000000000000000
```

In gem5, it prints:
```
0000000000000080
0000000000000000
00000000000000ff
000000000000f000
0000000000000000
0000000000000000
000000000000ffff
0000000000000000
00000000000000ff
```

I.e., the output of the first and last tests are incorrect.

**Affects version**
develop @ [86e08e8](https://github.com/gem5/gem5/commit/86e08e87d3917602db2fef0b8052df61237cc2ab)

**gem5 Modifications**
None

**To Reproduce**
1. Compile gem: `scons build/X86/gem5.opt`
2. Compile the test program: `cc -o test test.c`
3. Run the test program on the host: `./test > host.txt`
4. Run the test program in gem5: `./build/X86/gem5.opt configs/deprecated/example/se.py --cmd ./test --output=$PWD/sim.txt`
5. Observe the differences in output: `diff host.txt sim.txt`

**Terminal Output**
```
$ diff host.txt sim.txt
3c3
< 0000000000000080
---
> 00000000000000ff
9c9
< 0000000000000000
---
> 00000000000000ff
```

**Expected behavior**
The program should print the same output in gem5 as on the host.

**Host Operating System**
Ubuntu 22.04

**Host ISA**
X86

**Compiler used**
GCC 11.4.0

**Additional information**

nmosier, 2025-02-03:
Actually, it looks like my patch #1952 didn't fix all the bugs. Here's an updated test program that demonstrates another discrepancy between host vs. gem5 outputs:
```c
#include <stdio.h>
#include <stdint.h>

static uint64_t test(uint64_t a, uint64_t b) {
  uint64_t c;
  asm volatile ("movq %1, %%mm0\n"
                "movq %2, %%mm1\n"
                "packuswb %%mm1, %%mm0\n"
                "movq %%mm0, %0\n"
                : "=r"(c)
                : "r"(a), "r"(b));
  return c;
}

static void do_test(uint64_t a, uint64_t b) {
  const uint64_t c = test(a, b);
  printf("%016lx\n"
         "%016lx\n"
         "%016lx\n",
         a, b, c);
}

int main() {
  do_test(0x80, 0x80);
  do_test(0xF000, 0xF000);
  do_test(0xFFFF, 0xFFFF);
}
```

The host outputs:
```
0000000000000080
0000000000000080
0000008000000080
000000000000f000
000000000000f000
0000000000000000
000000000000ffff
000000000000ffff
0000000000000000
```
whereas gem5 outputs:
```
0000000000000080
0000000000000080
000000ff00000080
000000000000f000
000000000000f000
0000000000000000
000000000000ffff
000000000000ffff
000000ff00000000
```

The problem is that there is duplicated code in the implementation of the `Pack` micro-op for handling the first vector register (in this case, `mm0`) and the second vector register (in this case, `mm1`). I only applied by fix to the first block of duplicated code. To fix the above bug, we can just apply my fix to the second block of duplicated code.

erin-le, 2025-03-27:
@nmosier Hello, thank you for your contributions and sorry for the delayed response. Would it be possible for you to submit a PR with the fix applied to the second block of code?

PR #1952 (2025-01-28): arch-x86: fix pack micro-op implementation
Fix #1951. This patch fixes two bugs in the pack x86 micro-op implementation. The pack micro-op packs signed 2n-bit inputs into signed or unsigned n-bit outputs with saturation, i.e., if the 2n-bit value is out of the range that can be represented by the signed or unsigned n-bit value, it is replaced with -128/127 or 0/255.

Bug 1: a signed 2n-bit input that fits into one signed byte is not clamped to 0 before being copied to an unsigned n-bit output. This is because the checks when the input is negative were performed out of order.

Bug 2: a signed 2 [... cut]
commit 17f4c844 (2025-01-28): arch-x86: fix pack micro-op implementation (#1952)

Fix #1951. This patch fixes two bugs in the pack x86 micro-op
implementation. The pack micro-op packs signed 2n-bit inputs into signed
or unsigned n-bit outputs with saturation, i.e., if the 2n-bit value is
out of the range that can be represented by the signed or unsigned n-bit
value, it is replaced with -128/127 or 0/255.

Bug 1: a signed 2n-bit input that fits into one signed byte is not
clamped to 0 before being copied to an unsigned n-bit output. This is
because the checks when the input is negative were performed out of
order.

Bug 2: a [... cut]
closed 2025-04-28 by erin-le