gem5/gem5 #1956: Incorrect Summation in __riscv_vfredosum_vs_f64m2_f64m1 (LMUL>1) on gem5

ntampouratzis, 2025-01-29:
**Describe the bug**
I have the following simple c code which accumulates an array of 8 doubles (using vfloat64m2_t - (LMUL>1)).

```c
#include <stdio.h>
#include <riscv_vector.h>

int main() {
    // Initialize array with given values
    double arr[8] = {1, 4, 9, 16, 25, 36, 49, 64};
    
    // Load the array into a vector register
    vfloat64m2_t vec = __riscv_vle64_v_f64m2(arr, 8);
    
    // Temporary Vector
    vfloat64m1_t tmp = __riscv_vfmv_s_f_f64m1(0.0, 1);  // Initialize vector with 0.0
    
    // Perform the reduction sum operation
    vfloat64m1_t sum_vec = __riscv_vfredosum_vs_f64m2_f64m1(vec, tmp, 8);
    
    // Extract the final scalar sum result
    double sum = __riscv_vfmv_f_s_f64m1_f64(sum_vec);
    
    // Print the result
    printf("Sum: %lf\n", sum);
    
    return 0;
}
```

When executing the above simple RISC-V Vector (RVV 1.0) reduction sum operation using __riscv_vfredosum_vs_f64m2_f64m1 on gem5, the final sum result is incorrect. The expected result is 204.000000, but gem5 produces 174.000000. However, the same code executes correctly on QEMU, producing the expected sum.

**Affects version**
Latest gem5 version 24.1.0.1

**To Reproduce**
Compile and execute the above example on QEMU using:

```shell
gcc -O3 -march=rv64gcv -mabi=lp64d sum_rvv.c
```

**Terminal Output**
Run the program on QEMU (expected output):
```shell
Sum: 204.000000
```

Run the program on GEM5(expected output):
```shell
Sum: 174.000000
```

**Observed Behavior in gem5:**
The reduction sum operation incorrectly overrides the sum from the first vector register instead of adding it to the sum of the first register.
This indicates that __riscv_vfredosum_vs_f64m2_f64m1 is not correctly handling elements from both vector registers in vfloat64m2_t.

**Expected behavior**
The function should correctly accumulate the sum of all 8 elements across both registers.
Since vfloat64m2_t consists of two registers, gem5 should correctly process elements across them rather than discarding or overwriting values from the first register.

**Host Operating System**
Ubuntu 22.04

**Host ISA**
X-86.

**Additional information**
The incorrect summation occurs in all vfredosum which uses LMUL>1 (i.e. __riscv_vfredosum_vs_f64m4_f64m1, __riscv_vfredosum_vs_f64m8_f64m1).  In addition, the incorrect summation occurs in all vfredusum which uses LMUL>1.

To be noticed that the __riscv_vfredosum_vs_f64m1_f64m1 (LMUL=1) working properly on gem5.

saul44203, 2025-01-29:
Hey @ntampouratzis, #1955 should fix this.

ntampouratzis, 2025-01-30:
Dear @saul44203, yes it is working properly now! Thank you!! You may close the ticket!

closed 2025-01-30 by powerjg