gem5/gem5 #1960: stdlib: add SE mode support to multi-program workloads

sangjae4309, 2025-01-31:
#### Describe the enhancement
The current stdlib does not support multi-program workloads, which were previously supported by the now-deprecated se.py. In memory architecture research, inducing memory congestion is crucial, necessitating support for multi-program workloads

#### Implementation
```python
board.set_se_binary_workload(
    binary=[BinaryResource("./src_print_hello/print-hello"),
    BinaryResource("./src_print_hello/print-hello"),
    BinaryResource("./src_matrix_mul/matrix-mul"),
    BinaryResource("./src_matrix_mul/matrix-mul"),
    ],
    arguments=[[f"argument1", 1],
        [f"argument2", 1],
        [],
        [],
    ],
)
```
The function `set_se_binary_workload()` is now compatible with list. If a list is provided, the function distributes each binaries and its corresponding argument to respective core. There are no changes on single-program case.

There is a few consideration too:
  - The number of programs to be run shoubled match the number of cores (`se.py` was same before)
  - Mixing multi-thread program can produce unexpected result.

powerjg, 2025-01-31:
This is a good idea! I would suggest adding a new function (e.g., `set_se_multi_binary_workload`) to reduce the complexity of the original function. 

One important detail that may make this more difficult is how to integrate it with the `Workload` in gem5 Resources. I don't think the `Workload` object will support this out-of-the-box. However, it is a cool thing to try to implement.

RashidAGP, 2025-02-13:
I have done this before for a chiplet-based architecture, where each chiplet can run a different program (with separate memory spaces) or each core within a chiplet runs a different program. However, this approach can also be easily applied to a traditional architecture.

For this, you should assign `multiprocesses` variables to different `system.cpu` instances in `se.py`. Let me know if you need any guidance on how to do this. Or I can share my code with you.

erin-le, 2025-03-05:
Closing as complete as the linked PR has been merged.

closed 2025-03-05 by erin-le