gem5/gem5 #1966: Potential invalid coherence handling in MESI_two_level (and MESI_Three_Level)

gh1403, 2025-02-02:
Hi,

I'm wondering if the LLC handling logic for L1_GETS and L1_GETX might be mixed up here

https://github.com/gem5/gem5/blob/c9625ce9cc5b5a90a38327de5ac0e1870974af5e/src/mem/ruby/protocol/MESI_Two_Level-L2cache.sm#L956 

https://github.com/gem5/gem5/blob/c9625ce9cc5b5a90a38327de5ac0e1870974af5e/src/mem/ruby/protocol/MESI_Two_Level-L2cache.sm#L971

Why does the LLC send back exclusive data in response to a GETS request rather than a GETX request?

Thanks

powerjg, 2025-02-04:
I'm pretty sure the protocol is correct. The confusion is the "Exclusive" means *clean* (i.e., the data is not dirty with respect to the L2 cache) in this case.

gh1403, 2025-02-04:
I see. I realized that the L1 handles these response packets correctly. Thanks! 😊

closed 2025-02-04 by gh1403