gem5/gem5 #1967: util-docker: missing gfortran on gcn-gpu docker build

clemdiep, 2025-02-03:
**Describe the bug**
The `gfrotran` compiler seems to be missing from the `gcn-gpu` docker image. This prevent the image to be built.

**Affects version**
v24.1

**gem5 Modifications**
None

**To Reproduce**
1. Execute:
```shell
cd util/dockerfiles/gcn-gpu/
docker build -t gem5-gcn-gpu .
```
2. Wait until the error occurs.

**Terminal Output**
Here is the revelant part for the error:

```shell
27.80 + cmake -DROCM_PATH=/opt/rocm -lpthread -DTensile_LOGIC=asm_full -DTensile_ARCHITECTURE=all -DTensile_CODE_OBJECT_VERSION=V3 -DCMAKE_BUILD_TYPE=Release -DBUILD_WITH_TENSILE_HOST=ON -DTensile_LIBRARY_FORMAT=msgpack -DRUN_HEADER_TESTING=OFF -DTensile_COMPILER=hipcc -DCPACK_SET_DESTDIR=OFF -DCMAKE_INSTALL_PREFIX=rocblas-install -DCPACK_PACKAGING_INSTALL_PREFIX=/opt/rocm ../..
28.69 -- The CXX compiler identification is Clang 12.0.0
28.70 -- The Fortran compiler identification is unknown
28.71 -- Check for working CXX compiler: /opt/rocm/bin/hipcc
29.77 -- Check for working CXX compiler: /opt/rocm/bin/hipcc -- works
29.77 -- Detecting CXX compiler ABI info
30.82 -- Detecting CXX compiler ABI info - done
30.83 -- Detecting CXX compile features
30.83 -- Detecting CXX compile features - done
30.83 CMake Error at CMakeLists.txt:31 (project):
30.83   The CMAKE_Fortran_COMPILER:
30.83 
30.83     gfortran
30.83 
30.83   is not a full path and was not found in the PATH.
30.83 
30.83   Tell CMake where to find the compiler by setting either the environment
30.83   variable "FC" or the CMake cache entry CMAKE_Fortran_COMPILER to the full
30.83   path to the compiler, or to the compiler name if it is in the PATH.
30.83 
30.83 
30.83 -- Configuring incomplete, errors occurred!
30.83 See also "/rocBLAS/build/release/CMakeFiles/CMakeOutput.log".
30.83 See also "/rocBLAS/build/release/CMakeFiles/CMakeError.log".
30.83 + check_exit_code 1
30.83 + ((  1 != 0  ))
30.83 + exit 1
------
Dockerfile:156
```

**Expected behavior**
The docker image to build successfully.

**Host Operating System**
Ubuntu 24.04 LTS

**Host ISA**
X86

**Compiler used**
Docker version 27.5.1

**Additional information**
Adding rhe package `gfortran` in the `apt install` list, the problem is solved. If this is the wanted fix, I can make a PR with it.

mattsinc, 2025-02-03:
@clemdiep : can you please tell us more about what you are trying to run?  We didn't have gfortran in the docker because no applications used it.  So I'm not opposed to adding it, but it does make things larger and if there aren't workloads using it, then I guess we'd need to discuss with @BobbyRBruce and @powerjg if this is better to have in the mainline, or for you to have locally.

clemdiep, 2025-02-04:
> can you please tell us more about what you are trying to run?

I was trying to build the docker image as explained in the documentation [System Emulation AMD VEGA GPU model](https://www.gem5.org/documentation/general_docs/gpu_models/vega) and it fails when building/installing `rocBLAS`. I made no modification and don't want to do anything special.

I updated the original post to complete the terminal output with the actual Docker command that fails.

mattsinc, 2025-02-04:
Interesting, I have never encountered this before ... you are saying when you run docker pull ghcr.io/gem5-test/gcn-gpu:v23-1 it fails?

mattsinc, 2025-02-04:
Nevermind, I see in your updated post you are having docker build fail.  Very strange.

@abmerop have you ever seen this?  I'm not opposed to adding gfortran, but I also cannot replicate this issue thus far ...

mattsinc, 2025-02-06:
@abmerop thoughts?

mattsinc, 2025-02-06:
> It seems like one of the top level apt package dependencies changed, so I am OK to add this. I cannot re-create the docker image though.
> 
> I am also obligated to recommend using full-system mode instead :)

Thanks ... @clemdiep please just create a patch then

closed 2025-02-14 by clemdiep