gem5/gem5 #1986: the lack of algorithm header in random.cc

omerkarslioglu, 2025-02-08:
When I try to build latest version of gemv I got that error:

```
src/base/random.cc:59:27: error: ‘remove_if’ is not a member of ‘std’; did you mean ‘remove_cv’?
   59 |     instances->erase(std::remove_if(instances->begin(), instances->end(),
      |                           ^~~~~~~~~
      |                           remove_cv
 [ TRACING]  -> RISCV/debug/Decode.cc
```

I fixed with adding ``#include <algorithm>`` just in random.cc

**Affects version**
https://github.com/gem5/gem5/pull/1875

**To Reproduce**
Build gemv with:

```
scons build/RISCV/gem5.opt -j {cpus}
```

**Expected behavior**
You are gonna get an error on Fedora.

**Host Operating System**
Fedora Linux 41

**Host ISA**
RISC-V

adambagley, 2025-02-12:
FYI I had same issue on Ubuntu 24.10, and the suggested fix of adding `#include <algorithm>` worked for me as well. Thank you @omerkarslioglu !

erin-le, 2025-02-13:
@omerkarslioglu: Hello, thank you for reporting this issue! Would you be willing to submit a PR with this fix? 

[Here](https://bootcamp.gem5.org/#06-Contributing/01-contributing) is a guide on contribution to gem5. In particular, pages 32-33 explain how to format a commit message, and pages 29-31 contain information about pre-commit, which is a formatting tool.

omerkarslioglu, 2025-02-13:
@erin-le thanks, I am gonna try to contribute

closed 2025-02-18 by omerkarslioglu

EricSongXinLe, 2025-03-02:
Bug replicated on Ubuntu 24. I'll try to file a PR to fix it.