gem5/gem5 #1996: Inconsistent Data Type Conversion in read System Call Causes Allocation Issues in SE Mode (MemoryError: std::bad_alloc)

RashidAGP, 2025-02-12:
**Hi**
When running large Kronecker graphs in SE mode, I encountered an error (_MemoryError: std::bad_alloc_) that occurs specifically when a system call is invoked by the CPU. After investigating the issue for several days, I identified an inconsistency in data type conversion within the read system call implementation.

In `syscall_emul_buf.hh`, the size parameter is defined as an `int`, whereas in the call to `buf_arg(buf_ptr, nbytes)`, `nbytes` is of type `size_t`. This type mismatch can lead to incorrect memory allocations or overflows when handling large buffer sizes.

**Proposed Fix:**
To ensure correctness and consistency, the type of `size` in `syscall_emul_buf.hh` should be changed from int to `size_t`. This modification will align it with the standard definition of read system calls and prevent potential issues when handling large data sizes.

Let me know if you need additional details or test cases to reproduce the issue.

giactra, 2025-02-12:
Thanks for handling this.

>  To ensure correctness and consistency, the type of size in syscall_emul_buf.hh should be changed from int to size_t. This modification will align it with the standard definition of read system calls and prevent potential issues when handling large data sizes.

The comment I have is that BaseBufferArg is not used by the read syscall only. Multiple syscalls whose size parameter is not necessarily size_t are relying on it. I believe there is no harm in being explicit and make the size parameter 64bit-wide for *every* host machine.

This is not the case for size_t as an example. So I vote for using either **uint64_t** or **Addr** itself

RashidAGP, 2025-02-13:
OK, good to know. I had it only in the `read` system call. But if you think other syscalls might also use different types, then it should be changed to `uint64_t`.

PR #1997 (2025-04-03): sim: Fix MemoryError at system call
This fixes an issue with inconsistent data type conversion in the `read` system call.  
Fixes #1996
commit 2c6cac9e (2025-04-03): sim: Fix MemoryError at system call (#1997)

This fixes an issue with inconsistent data type conversion in the `read`
system call.
Fixes #1996

---------
closed 2025-04-03 by erin-le