gem5/gem5 #2001: [arm][bug]panic: Attempted to execute unknown instruction (inst 0x6e79d884)

ChenBoya-prj, 2025-02-13:
**Describe the bug**

panic: Attempted to execute unknown instruction (inst 0x6e79d884))
I disassembled the binary, the instruction is 
6e79d884        ucvtf   v4.8h, v4.8h

**Affects version**
v24.1.0.1 Hotfix Release  (#1875)

**gem5 Modifications**
no modifications

**To Reproduce**
use simple SE mode to run

**Host Operating System**
 Ubuntu 18.04.6 LTS

**Host ISA**
X86

**Compiler used**
scons build/ARM/gem5.opt
gcc (GCC) 10.3.0

giactra, 2025-02-13:
This is because we don't currently support FEAT_FP16. I have an internal implementation of it I shall post soon

erin-le, 2025-03-20:
The linked PR has been merged into the `develop` branch, so this issue is being closed as completed. Please feel free to reopen if the issue still occurs.

closed 2025-03-20 by erin-le