gem5/gem5 #2008: Unable to build any gem5 binary, regardless of ISA or type

lusinge, 2025-02-15:
**Describe the bug**
I am using the official docker image (24.04, all deps) on a unmodified `stable` branch. When I try to compile with any number of processing units, I get the following error :

```shell
 [ENUMDECL] m5.objects.Graphics, ImageFormat -> RISCV/enums/ImageFormat.hh
 [SO Param] m5.objects.CPUTracers, IntelTrace -> RISCV/python/_m5/param_IntelTrace.cc
terminate called after throwing an instance of 'pybind11::error_already_set'
terminate called recursively
terminate called after throwing an instance of 'pybind11::error_already_set'
terminate called recursively
Aborted (core dumped)
scons: *** [build/RISCV/enums/ImageFormat.hh] Error 134
Aborted (core dumped)
scons: *** [build/RISCV/python/_m5/param_IntelTrace.cc] Error 134
scons: building terminated because of errors.
```

The file that causes the error changes everytime I try to compile gem5. Once I get this error, scons refuses to compile any further.

The same 'pybind11::error_already_set' error will appear regardless of compiling a RISCV .fast, a X86 .opt, a X86 .fast...

**Affects version**
24.1.0.2

**gem5 Modifications**
None

**To Reproduce**
Compile gem5 from its unmodified source using the official docker image (24.04, all deps).

**Terminal Output**

```shell
$ scons build/RISCV/gem5.opt -j$(nproc --ignore=4)
scons: Reading SConscript files ...

You're missing the pre-commit/commit-msg hooks. These hook help to ensure your
code follows gem5's style rules on git commit and your commit messages follow
our commit message requirements. This script will now install these hooks in
your .git/hooks/ directory.
Press enter to continue, or ctrl-c to abort:
 
pre-commit installed at .git/hooks/pre-commit
pre-commit installed at .git/hooks/commit-msg
Mkdir("/gem5/build/RISCV/gem5.build")
Checking for linker -Wl,--as-needed support... yes
Checking for compiler -gz support... yes
Checking for linker -gz support... yes
Info: Using Python config: python3-config
Checking for C header file Python.h... yes
#[...]
Checking for backtrace_symbols_fd((void *)1, 0, 0) in C library None... yes
"ext/Kconfiglib/defconfig.py" --kconfig "/gem5/build/RISCV/gem5.build/Kconfig" "/gem5/build_opts/RISCV"
Loaded configuration '/gem5/build_opts/RISCV'
Configuration saved to '/gem5/build/RISCV/gem5.build/config'
Checking whether __i386__ is declared... no
Checking whether __x86_64__ is declared... yes
Generating LALR tables
WARNING: 4 shift/reduce conflicts
WARNING: 1 reduce/reduce conflict
WARNING: reduce/reduce conflict in state 131 resolved using rule (params -> empty)
WARNING: rejected rule (types -> empty) in state 131
BUILD_TLM not set, not building CHI-TLM integration

Checking for compiler -Wno-self-assign-overloaded support... yes
Checking for linker -Wno-free-nonheap-object support... yes
scons: done reading SConscript files.
scons: Building targets ...
 [     CXX] src/python/gem5py.cc -> RISCV/python/gem5py.pyo
 [CONFIG H] HAVE_DEPRECATED_NAMESPACE, 1 -> RISCV/config/have_deprecated_namespace.hh
[... 43 lines cut ...]
terminate called after throwing an instance of 'pybind11::error_already_set'
terminate called recursively
terminate called after throwing an instance of 'pybind11::error_already_set'
terminate called recursively
Aborted (core dumped)
scons: *** [build/RISCV/enums/ImageFormat.hh] Error 134
Aborted (core dumped)
scons: *** [build/RISCV/python/_m5/param_IntelTrace.cc] Error 134
scons: building terminated because of errors.
```

**Expected behavior**
Build and obtain a functionning gem5.opt

**Host Operating System**
Ubuntu 24.04 container

**Host ISA**
X86

**Compiler used**
gem5's docker image compiler

**Additional information**
I don't think this is a hardware or memory issue. It's very hard to troubleshoot and I can't find any other issue with the same problem.

BobbyRBruce, 2025-02-17:
`Error 134` is typically returned by Docker because it's running out of memory. Either the Docker container isn't being allocated enough memory, or you don't have enough memory on your host generally for Docker to use.

lusinge, 2025-02-21:
Uh, weird. I actually used gem5 in a distrobox (basically podman container that uses the host's home as a volume path for the container's home) . When I use a real podman container, it works. So I don't know where, but distrobox must brick the container somehow. I'll close this as this not a gem5 issue. Thank you !

closed 2025-02-21 by lusinge