gem5/gem5 #2020: Can gem5 support modeling differen XCDs working together?

gty111, 2025-02-20:
Now gem5 support model a single "XCD" in MI300X. But MI300X includes 8 XCDs. So can gem5 support modeling 8XCDs working together?

abmerop, 2025-02-20:
If you want to add more XCDs and have a single command processor controlling them all, it is just a matter of changing the topology.

If you want to add more XCDs and have them be seen as independent devices, that is not yet possible, but is being worked on.

gty111, 2025-02-21:
Thanks for the reply. Do 8 XCDs in MI300 work indepently in fact? In the white paper, it says each XCD has its own hardware scheduler.

gty111, 2025-02-21:
> If you want to add more XCDs and have them be seen as independent devices, that is not yet possible, but is being worked on.

By the way, can you shed some light on the development progress ?

mattsinc, 2025-02-21:
The current support for multi-chiplets is a very old version of ROCm (and thus a very old version of gem5): https://github.com/hal-uw/cpelide-micro24-artifact

We are working on getting this working with the current head of ROCm and gem5, but no ETA on when it will be done -- @abmerop explained above what does work today.

In terms of MI300, as @abmerop alluded to, there are different modes that a (real) MI300 GPU can work in.  One of those modes is what you described.  However, currently in gem5 only what @abmerop explained works today.

gty111, 2025-02-21:
Thanks for the detailed explannation.

abmerop, 2025-02-22:
> Thanks for the reply. Do 8 XCDs in MI300 work indepently in fact? In the white paper, it says each XCD has its own hardware scheduler.

Yes, each "compute partition" has it's own command processor.  You could have one compute partition (current gem5 setup) and up to one compute partition per XCD. 

> By the way, can you shed some light on the development progress ?

Basically need to do the following:
1. Use real MI300X bios and firmware. This is done: https://github.com/abmerop/gem5/tree/mi300x
2. Update disk image to include said firmware. Also done: https://github.com/abmerop/gem5-resources/tree/rocm63
3. Rework hardware queues to be handled by CPs instead of the device: ~90% complete
4. [Optionally] Add support for memory partitions (called "NPS" or NUMAs Per Socket).

Possibly some other steps may come up while working on those steps.  I don't expect the support to be there until next gem5 major release. The changes in the first two steps will require the new disk image, so it makes the most sense to release them at the same time.