gem5/gem5 #2031: Incorrect Parameter in violation Function Call for Address Overlap Handling

gerayking, 2025-02-24:
First, I need to make sure I understand the code structure. The violation method is part of the instruction queue, and it's supposed to handle memory order violations. 
- violator is obtained via ldstQueue.getMemDepViolator(tid), which typically returns a store instruction that caused the memory order violation
- inst is the current load instruction being executed that detected the violation

I think the violator is the load instruction that incurs a violation, but the inst being executed should not be the first parameter of the violation function. The violation function requires both the load instruction and the store instruction whose addresses overlap.

https://github.com/gem5/gem5/blame/186a913a48f13bdd484fcdef17ac3e28d2b8b4c9/src/cpu/o3/iew.cc#L1297

Harshil2107, 2025-02-27:
Hi @gerayking,

Thank you for reporting this issue. I just want to clarify my understanding of the issue. It seems to me that you are saying that the order in which the parameters are being passed in the `instQueue.violation(inst, violator);` call might be incorrect, as `inst` is a load instruction and `violator` is a store instruction that caused the violation. According to the definition of the `violation` function:

```cpp
void InstructionQueue::violation(const DynInstPtr &store, const DynInstPtr &faulting_load)
```

Are you suggesting that we should be calling instQueue.violation(violator, inst); instead?

Could you provide an example where this issue is highlighted or any guidance on how to verify the fix for this issue?

Thank you.

erin-le, 2025-04-03:
@gerayking Hello, could you provide a binary that is executed incorrectly because of this issue?

closed 2025-05-02 by github-actions[bot]