gem5/gem5 #2072: zlib header file not found

davidcb98, 2025-03-11:
Hello,

I have a problem compiling and I get the error that the zlib.h is not found. 

I think the problem comes from the fact that the zlib.h is not in a standard folder of the operating system, since I have compiled this library in my Home. 

Normally what I do to pass the headers to a compiler is:
export CFLAGS=/home/....
export CPPFLAGS=/home/...
export CXXFLAGS=/home/...

The problem is that I think the scons config file ignores these shell definitions and redefines these variables. 

Any idea how to tell the compiler where my zlib.h is?

Thank you very much,
David

davidcb98, 2025-03-11:
I make update. It seems that it is not zlib.h.

I found this in log build/x86/gem5.build/scons_config.log

> #################################################################
> scons: Configure: Checking for C++ library z...
> 
> build/x86/gem5.build/scons_config/conftest_74b84a772eb7a0f5d1595222ad4fe30c_0.cpp <-
>   |
>   |
>   |#include "zlib.h"
>   |
>   |int main(void) {
>   |
>   |return 0;
>   |}
>   |
> 
>  [     CXX] x86/gem5.build/scons_config/conftest_74b84a772eb7a0f5d1595222ad4fe30c_0.cpp -> .o
>  [    LINK]  -> x86/gem5.build/scons_config/conftest_74b84a772eb7a0f5d1595222ad4fe30c_0_ed8a156d19010574e923df8add905d42
> 
> /localscratch/users/tmp/tmpqynl0jia: line 1: syntax error near unexpected token `('
> 
> /localscratch/users/tmp/tmpqynl0jia: line 1: `/mnt/home/soft/gcc21/programs/x86_64/gcc-13.2.0/bin/g++ -o "build/x86/gem5.build/scons_config/conftest_74b84a772eb7a0f5d1595222ad4fe30c_0_ed8a156d19010574e923df8add905d42" -z origin -Wl,--as-needed -gz "build/x86/gem5.build/scons_config/conftest_74b84a772eb7a0f5d1595222ad4fe30c_0.o" -L/mnt/home/soft/gem5/programs/x86_64/24.1.0.2_env/lib/python3.13/config-3.13-x86_64-linux-gnu -L/mnt/home/soft/gem5/programs/x86_64/24.1.0.2_env/lib -lpython3.13 -lpthread -ldl -lutil -lm -lz -lz -ll -li -lb -lV -le -lr -ls -li -lo -ln -l( -l) -l;'
> 
> scons: Configure: no
> ##################################################################

It seems that the problem is that the sconf generates the wrong statement to compile the program to verify the zlib.h. Specifically, all this is left over: 

"-lz -ll -li -lb -lV -le -lr -ls -li -lo -ln -l( -l) -l" 

Note that "-lz" is duplicated, so there is one left over.

If I execute that statement in terminal without the commented ones, it compiles. Come on, the commented, it is not the zlib.h. 

Any idea?

Thank you very much,
David

davidcb98, 2025-03-12:
I have found the problem. It seems that the software does not compile well with modern versions of scons. I was using version 4.9.0. I have switched to version 3.1.2 and it compiles fine.

powerjg, 2025-03-12:
What Linux distro/version are you using that has scons 4.9 installed?

davidcb98, 2025-03-13:
Good morning,

thank you very much for your reply. I use OpenSuse 15.4, but I installed scons in a conda environment:

```
conda create -p ~/programs/x86_64/24.1.0.2_env
source activate ~/programs/x86_64/24.1.0.2_env
conda install -c conda-forge python=3.11.4 scons==3.1.2 mypy==1.13.0 pre-commit==4.0.1 tqdm==4.66.6
export LD_LIBRARY_PATH=~/programs/x86_64/24.1.0.2_env/lib:${LD_LIBRARY_PATH}
```

Thank you very much,
David

powerjg, 2025-03-13:
conda/venv/poetry/etc. do not generally play well with gem5. I'm happy that you seem to have gotten it to work, though! 

We'll look into the scons 4.9 issue. Thanks for the information.

Harshil2107, 2025-03-14:
I have pushed a PR that should fix this issue.

PR #2080

closed 2025-03-19 by erin-le