gem5/gem5 #2073: Devcontainer Build Fails Due to Outdated GCC Version (9.4.0) - Requires GCC 10 or Newer

sigongzi, 2025-03-13:
**Describe the bug**  
When attempting to build gem5 using the provided devcontainer image, the build process fails with an error indicating that the GCC version is too old. The error message states that GCC version 10 or newer is required, but the installed version in the devcontainer is 9.4.0. Additionally, there is a warning about missing `pre-commit` hooks due to unmet Python requirements, though I chose to continue the compilation.

**Affects version**  
This issue was encountered while using the gem5 repository in the `develop` branch.

**gem5 Modifications**  
No modifications were made to the gem5 source code. I am using the repository as-is within the devcontainer environment provided by the gem5 repository.

**To Reproduce**  
Steps to reproduce the behavior, starting from a clean repository:

1. Clone the gem5 repository and open it in a devcontainer environment (using the provided `.devcontainer` configuration in the gem5 repository).
2. Inside the devcontainer, attempt to compile gem5 with the following command:  
   ```shell
   scons build/RISCV/gem5.opt
   ```
3. When prompted about missing `pre-commit` hooks, press `y` to continue compilation.

**Terminal Output**  
The relevant terminal output from the failed build attempt is as follows:  

```shell
scons: Reading SConscript files ...

You're missing the pre-commit/commit-msg hooks. These hook help to ensure your
code follows gem5's style rules on git commit and your commit messages follow
our commit message requirements. This script will now install these hooks in
your .git/hooks/ directory.
Press enter to continue, or ctrl-c to abort:
 Cannot find 'pre-commit'. Please ensure all Python requirements are 
installed. This can be done via 'pip install -r requirements.txt'.
It is strongly recommended you install the pre-commit hooks before working with gem5. Do you want to continue compilation (y/n)?
y
Mkdir("/workspaces/gem5/build/RISCV/gem5.build")
Checking for linker -Wl,--as-needed support... yes
Error: gcc version 10 or newer required.
       Installed version: 9.4.0
```

**Expected behavior**  
I expected the build process to complete successfully without errors related to the GCC version. The devcontainer image provided by the gem5 repository should ideally include a compatible version of GCC (version 10 or newer) to allow for a smooth build process.

**Host Operating System**  
The host operating system inside the devcontainer appears to be Ubuntu-based (likely Ubuntu 20.04, based on the GCC version 9.4.0). I can confirm the exact version if needed.

**Host ISA**  
The devcontainer environment is running on x86_64 architecture (assumed based on typical devcontainer setups). I am targeting the RISC-V ISA for the gem5 build (as indicated by the `build/RISCV/gem5.opt` target).

**Compiler used**  
The compiler used is GCC version 9.4.0, which is pre-installed in the devcontainer image provided by the gem5 repository. The build process requires GCC version 10 or newer.

**Additional information**  
- The devcontainer image provided by the gem5 repository seems to be outdated or misconfigured, as it includes GCC 9.4.0 instead of the required GCC 10 or newer.
- The warning about missing `pre-commit` hooks suggests that the Python dependencies (e.g., `pre-commit`) are not installed in the devcontainer by default. While this did not directly cause the build failure, it might be worth addressing in the devcontainer setup for a smoother development experience.
- To resolve this issue, it may be necessary to update the devcontainer image to include a newer version of GCC (e.g., GCC 11 or 12) and ensure that all Python dependencies are pre-installed.

powerjg, 2025-03-13:
What version of gem5 are you using?

sigongzi, 2025-03-17:
v24.1.0.2

BobbyRBruce, 2025-04-07:
I'm struggling to recreate this issue here. When I get `docker pull ghcr.io/gem5/devcontainer:latest` or `docker pull ghcr.io/gem5/devcontainer:v24-1` and spin up containers with them I see gcc >10 installed in each case (`docker run -u$UID:$GID -it --rm ghcr.io/gem5/devcontainer:latest gcc -v`).

Can you give me steps you used to try and recreate this issue?

sigongzi, 2025-04-08:
I previously had this issue. I cloned the repo with git clone https://github.com/gem5/gem5.git and opened it in a devcontainer via VS Code's "Dev Container: Open Folder in Container". There were GCC version and pre-commit hook problems. But when I retried today, the issue disappeared. Not sure why. Maybe it's due to devcontainer or system updates.