gem5/gem5 #2109: [v24.1.0.2] Invalid range for type MachineType

pengzhangpk, 2025-03-24:
I compiled gem5 with scons build 'scons build/ARM/gem5.fast -j40' on ubuntu 2204 with x86_64 platform. When I run cmd './build/ARM/gem5.fast configs/example/ruby_random_test.py -n 2' , the error occurred: 
panic: Invalid range for type MachineType
```txt
gem5 version 24.1.0.2
gem5 compiled Mar 24 2025 16:52:29
gem5 started Mar 24 2025 20:31:30
gem5 executing on iZbp197y9af88669d1to18Z, pid 239971
command line: ./build/ARM/gem5.fast configs/example/ruby_random_test.py -n 2

Global frequency set at 1000000000 ticks per second
warn: system.ruby.network.ext_links0.ext_node adopting orphan SimObject param 'sequencer'
warn: system.ruby.network.ext_links1.ext_node adopting orphan SimObject param 'sequencer'
warn: rounding error > tolerance
    1.250000 rounded to 1
warn: rounding error > tolerance
    1.250000 rounded to 1
warn: rounding error > tolerance
    1.250000 rounded to 1
warn: rounding error > tolerance
    1.250000 rounded to 1
warn: rounding error > tolerance
    1.250000 rounded to 1
warn: rounding error > tolerance
    1.250000 rounded to 1
warn: failed to generate dot output from m5out/config.dot
warn: failed to generate dot output from m5out/config.system.ruby.dot
warn: rounding error > tolerance
    1.250000 rounded to 1
warn: rounding error > tolerance
    1.250000 rounded to 1
src/mem/dram_interface.cc:690: warn: DRAM device capacity (8192 Mbytes) does not match the address range assigned (512 Mbytes)
src/base/statistics.hh:279: warn: One of the stats is a legacy stat. Legacy stat is a stat that does not belong to any statistics::Grou. Legacy stat is deprecated.
src/sim/simulate.cc:199: info: Entering event queue @ 0.  Starting simulation...
build/ARM/mem/ruby/protocol/MachineType.cc:77: panic: Invalid range for type MachineType
Memory Usage: 730648 KBytes
Program aborted at tick 4
--- BEGIN LIBC BACKTRACE ---
./build/ARM/gem5.fast(_ZN4gem515print_backtraceEv+0x30)[0x562b6a33c290]
./build/ARM/gem5.fast(_ZN4gem512abortHandlerEi+0x4c)[0x562b6a360f9c]
/lib/x86_64-linux-gnu/libc.so.6(+0x42520)[0x7fa52d9a6520]
/lib/x86_64-linux-gnu/libc.so.6(pthread_kill+0x12c)[0x7fa52d9fa9fc]
/lib/x86_64-linux-gnu/libc.so.6(raise+0x16)[0x7fa52d9a6476]
/lib/x86_64-linux-gnu/libc.so.6(abort+0xd3)[0x7fa52d98c7f3]
./build/ARM/gem5.fast(+0xffb675)[0x562b6a34e675]
./build/ARM/gem5.fast(_ZN4gem54ruby21MachineType_to_stringB5cxx11ERKNS0_11MachineTypeE+0x416)[0x562b6bdb4ad6]
./build/ARM/gem5.fast(_ZN4gem54rubylsERSoRKNS0_11MachineTypeE+0x25)[0x562b6bdb4b05]
./build/ARM/gem5.fast(+0x2b2e2ed)[0x562b6be812ed]
./build/ARM/gem5.fast(_ZN4gem58ccprintfImJNS_4ruby11MachineTypeEEEEvRNS_2cp5PrintERKT_DpRKT0_+0x3c7)[0x562b6be82fc7]
./build/ARM/gem5.fast(_ZN4gem56Logger5printIJNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEmNS_4ruby11MachineTypeEEEEvRKNS0_3LocEKcDpRKT_+0x1fa)[0x562b6be8338a]
./build/ARM/gem5.fast(_ZNK4gem54ruby18AbstractController29mapAddressToDownstreamMachineEmNS0_11MachineTypeE+0x2cc)[0x562b6be7f25c]
./build/ARM/gem5.fast(_ZN4gem54ruby3CHI16Cache_Controller22checkResourceAvailableERKNS1_17Cache_RequestTypeERKm+0xe0)[0x562b6bdc55f0]
./build/ARM/gem5.fast(_ZN4gem54ruby3CHI16Cache_Controller18doTransitionWorkerENS1_11Cache_EventENS1_11Cache_StateERS4_RPNS1_9Cache_TBEEPNS1_16Cache_CacheEntryEm+0x6853)[0x562b6bd82e63]
./build/ARM/gem5.fast(_ZN4gem54ruby3CHI16Cache_Controller12doTransitionENS1_11Cache_EventEPNS1_16Cache_CacheEntryEPNS1_9Cache_TBEEm+0x7)[0x562b6bd840e5]
./build/ARM/gem5.fast(_ZN4gem54ruby3CHI16Cache_Controller6wakeupEv+0xd92)[0x562b6bd970e2]
./build/ARM/gem5.fast(+0x2b1eb21)[0x562b6be71b21]
./build/ARM/gem5.fast(_ZN4gem510EventQueue10serviceOneEv+0xa8)[0x562b6a350258]
./build/ARM/gem5.fast(_ZN4gem59doSimLoopEPNS_10EventQueueE+0x68)[0x562b6a388768]
./build/ARM/gem5.fast(_ZN4gem58simulateEm+0x25b)[0x562b6a388cdb]
./build/ARM/gem5.fast(+0x26fafb0)[0x562b6ba4dfb0]
./build/ARM/gem5.fast(+0xf8a553)[0x562b6a2dd553]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x128023)[0x7fa52e4a4023]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyObject_Call+0x5c)[0x7fa52e45dfec]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x4b16)[0x7fa52e3f2776]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af)[0x7fa52e53e3af]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9d68)[0x7fa52e3f79c8]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af)[0x7fa52e53e3af]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(PyEval_EvalCode+0xbe)[0x7fa52e5393de]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1bd96d)[0x7fa52e53996d]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1287b3)[0x7fa52e4a47b3]
--- END LIBC BACKTRACE ---
For more info on how to address this issue, please visit https://www.gem5.org/documentation/general_docs/common-errors/ 

Aborted (core dumped)
```

powerjg, 2025-04-08:
The Ruby random tester is not compatible with the CHI protocol by using `ruby_random_test.py` and `Ruby.py`. It is not simple or straightforward to make these scripts compatible.

The main problem is that the `CHI_config.py` assumes that the CPU list passed in (to the `CHI_RNF.generate` method) is a list of actual CPUs. However, `ruby_random_test.py` tries to trick the generic Ruby protocol configuration files into thinking that a *single* tester is actually multiple CPUs.

You can write a new run script from scratch that uses the `RubyTester` object with the CHI protocol, but we don't have any examples that are working out-of-the-box.

Another option is to use the `TrafficGenerator` objects. These are compatible with the stdlib (see `configs/example/gem5_library/memory_traffic.py` for an example). 

In general, we are deprecating all of the scripts in `configs/ruby` with preference to migrate things to the stdlib. `configs/ruby` is incompatible with the new support for multiple Ruby protocols, and it is not well maintained.

closed 2025-04-08 by powerjg

powerjg, 2025-04-08:
Note: #2160 makes it possible to use the random tester with the `ALL` gem5 build (except for CHI and the token protocol)

pengzhangpk, 2025-04-10:
@powerjg Thank you very much. I got to know it.