gem5/gem5 #2116: arch-arm: Linux kernel 6.8.12 causes ARM simulation to stall due to executing unkown intruction

Harshil2107, 2025-03-25:
#### **Describe the Bug**
When booting a Linux 6.8.12 kernel (built with the default configuration) on ARM in gem5 using **TIMING mode**, the simulation crashes during early EL2 initialization at the instruction `msr hcrx_el2, x0`.

At this point:
- Execution jumps to an invalid address `0x400`
- An undefined instruction (`0x000000`) is executed
- The simulation enters an infinite fault loop at `0x200`

This issue **only occurs in TIMING mode**. The same kernel and disk image boot successfully to completion in **KVM mode**.

---

#### **Affected Versions**
- 24.1  
- develop  

---

#### **To Reproduce**
1. Download the kernel and disk image:  
  [arm-kernel-disk.tar.gz](https://drive.google.com/file/d/1lkWUhLxjoJOWroVNuIhiyxqkk0f0cyRY/view?usp=sharing)

2. Build gem5:
   ```bash
   scons build/ALL/gem5.opt -j$(nproc)
   ```

3. Run with the following configuration (`config.py`):
   ```python
   import argparse
   from gem5.isas import ISA
   from gem5.prebuilt.demo.arm_demo_board import ArmDemoBoard
   from gem5.resources.resource import obtain_resource, KernelResource, DiskImageResource
   from gem5.simulate.simulator import Simulator
   from gem5.utils.requires import requires

   requires(isa_required=ISA.ARM)

   parser = argparse.ArgumentParser(description="Run ArmDemoBoard in gem5")
   parser.add_argument("--use-kvm", action="store_true", help="Use KVM cores")
   args = parser.parse_args()

   board = ArmDemoBoard(use_kvm=args.use_kvm)
   board.set_kernel_disk_workload(
       kernel=KernelResource("path/to/kernel"),
       disk_image=DiskImageResource("path/to/disk", root_partition="2"),
       bootloader=obtain_resource("arm64-bootloader-foundation")
   )

   simulator = Simulator(board=board)
   simulator.run()
   ```

4. Launch the simulation:
   ```bash
   ./build/ALL/gem5.opt -d m5out/arm-demo-run -re --debug-flag=Exec,Decode,Fetch config.py
   ```

**Terminal Output**
This is the relevent section from the output when the simulation is run with `--debug-flag=Exec,Decode,Fetch`

```shell
0355887: board.processor.cores0.core: T0 : 0x81b7c17c @kernel_init.init_el2+100    :   cbz   x0, <kernel_init.init_el2+116> : IntAlu : 
90355887: board.processor.cores0.core: Fetch: Inst PC:0x81b7c180, Fetch PC:0x81b7c180
90402507: board.processor.cores0.core.decoder: Decode: Decoded movz instruction: 0x4d2a00000
90402507: board.processor.cores0.core: T0 : 0x81b7c180 @kernel_init.init_el2+104    :   movz   x0, #0, #16       : IntAlu :  D=0x0000000000000000
90402507: board.processor.cores0.core: Fetch: Inst PC:0x81b7c184, Fetch PC:0x81b7c184
90449127: board.processor.cores0.core.decoder: Decode: Decoded movk instruction: 0x4f2890000
90449127: board.processor.cores0.core: T0 : 0x81b7c184 @kernel_init.init_el2+108    :   movk   x0, #18432, #0    : IntAlu :  D=0x0000000000004800
90449127: board.processor.cores0.core: Fetch: Inst PC:0x81b7c188, Fetch PC:0x81b7c188
90495747: board.processor.cores0.core.decoder: Decode: Decoded msr instruction: 0x4d51c1240
90495747: board.processor.cores0.core: T0 : 0x81b7c188 @kernel_init.init_el2+112    :   msr   hcrx_el2, x0       : IntAlu : 
90495747: board.processor.cores0.core: Fetch: Inst PC:0x000400, Fetch PC:0x000400
90535041: board.processor.cores0.core.decoder: Decode: Decoded unknown instruction: 0x400000000
90535041: board.processor.cores0.core: T0 : 0x400 @PECOFF_FILE_ALIGNMENT+512    : unknown    (inst 0x000000) : IntAlu : 
90535041: board.processor.cores0.core: Fetch: Inst PC:0x000200, Fetch PC:0x000200
90574335: board.processor.cores0.core.decoder: Decode: Decoded unknown instruction: 0x400000000
90574335: board.processor.cores0.core: T0 : 0x200 @PECOFF_FILE_ALIGNMENT    : unknown    (inst 0x000000) : IntAlu : 
90574335: board.processor.cores0.core: Fetch: Inst PC:0x000200, Fetch PC:0x000200
90613629: board.processor.cores0.core.decoder: Decode: Decoded unknown instruction: 0x400000000
90613629: board.processor.cores0.core: T0 : 0x200 @PECOFF_FILE_ALIGNMENT    : unknown    (inst 0x000000) : IntAlu : 
90613629: board.processor.cores0.core: Fetch: Inst PC:0x000200, Fetch PC:0x000200
90652923: board.processor.cores0.core.decoder: Decode: Decoded unknown instruction: 0x400000000
90652923: board.processor.cores0.core: T0 : 0x200 @PECOFF_FILE_ALIGNMENT    : unknown    (inst 0x000000) : IntAlu : 
90652923: board.processor.cores0.core: Fetch: Inst PC:0x000200, Fetch PC:0x000200
```

**Expected behavior**
The workload to boot to completion with all 3 hypercalls triggered.
**Host Operating System**
Ubuntu 24.04

**Host ISA**
X86

giactra, 2025-03-26:
This is most likely caused by the bootloader not setting the SCR_EL3.HXEN bit to 1.

https://github.com/gem5/gem5/blob/stable/src/arch/arm/regs/misc.cc#L2046
https://github.com/gem5/gem5/blob/stable/system/arm/bootloader/arm64/boot.S#L64

We should amend the bootloader and re-upload it

Harshil2107, 2025-03-27:
Thank you for pointing out the issue. Could you please direct me to the documentation on building or source for the bootloaders? I wasn't able to find the source in the gem5-resources repository.

PR #2130 (2025-04-01): arch-arm: Update bootloader to set SCR_EL3.HXEN bit to 1

This PR updates the ARM bootloader to set the `SCR_EL3.HXEN` bit (bit 38) to `1`.

### Details  
- Sets `SCR_EL3.HXEN` (bit 38) to `1` in the bootloader.  
- Aligns with ARM's specifications:  
  [ARM Documentation – SCR_EL3 Register](https://developer.arm.com/documentation/ddi0601/2025-03/AArch64-Registers/SCR-EL3--Secure-Configuration-Register)

### Fixes  
Fixes #2116.
commit e56775df (2025-04-01): arch-arm: Update bootloader to set SCR_EL3.HXEN bit to 1 (#2130)

This PR updates the ARM bootloader to set the `SCR_EL3.HXEN` bit (bit
38) to `1`.

### Details  
- Sets `SCR_EL3.HXEN` (bit 38) to `1` in the bootloader.  
- Aligns with ARM's specifications:  
[ARM Documentation – SCR_EL3
Register](https://developer.arm.com/documentation/ddi0601/2025-03/AArch64-Registers/SCR-EL3--Secure-Configuration-Register)

### Fixes  
Fixes #2116.
closed 2025-04-17 by Harshil2107