gem5/gem5 #2125: Segmentation Fault in gem5 Simulation with RVV VLEN > 512

ntampouratzis, 2025-03-28:
**Describe the bug**
When running a simple RVV-enabled matrix addition program using gem5 (latest version), the simulation crashes with a segmentation fault when VLEN is configured to be greater than 512. The code works correctly and shows performance improvement for VLEN <= 512, but fails for VLEN = 1024 or above.

**Affects version**
The latest stable version (24.1.0.2) in full system mode using the ./configs/example/gem5_library/riscv-ubuntu-run.py script with the provided Ubuntu 24.04 and provided 6.6.33 Linux kernel.

**To Reproduce**
1. Save the provided C code (below) as add_rvv.c
2. Compile it (gcc -O3 -march=rv64gcv -mabi=lp64d add_rvv.c)
3. Set VLEN=1024 or any value greater than 512 in gem5 file (line 100 of src/arch/riscv/RiscvISA.py)
4. Run the simulation and observe the crash.

```c
#include <stdio.h>
#include <stdlib.h>
#include <sys/time.h>
#include <riscv_vector.h>

#define ROWS 1024
#define COLS 1024

void serial_matrix_add(const float *A, const float *B, float *C, size_t rows, size_t cols) {
    size_t size = rows * cols;
    for (size_t i = 0; i < size; i++) {
        C[i] = A[i] + B[i];
    }
}

void vector_matrix_add(const float *A, const float *B, float *C, size_t rows, size_t cols) {
    size_t size = rows * cols;
    size_t vl;
    for (size_t i = 0; i < size; i += vl) {
        vl = __riscv_vsetvl_e32m1(size - i);

        vfloat32m1_t a = __riscv_vle32_v_f32m1(&A[i], vl);
        vfloat32m1_t b = __riscv_vle32_v_f32m1(&B[i], vl);
        vfloat32m1_t c = __riscv_vfadd_vv_f32m1(a, b, vl);
        __riscv_vse32_v_f32m1(&C[i], c, vl);
    }
}

double get_time_diff(struct timeval start, struct timeval end) {
    return (end.tv_sec - start.tv_sec) * 1e3 + (end.tv_usec - start.tv_usec) / 1e3;
}

int main() {
    size_t size = ROWS * COLS;
    float *A = (float *)malloc(size * sizeof(float));
    float *B = (float *)malloc(size * sizeof(float));
    float *C_serial = (float *)malloc(size * sizeof(float));
    float *C_vector = (float *)malloc(size * sizeof(float));

    // Initialize input matrices
    for (size_t i = 0; i < size; i++) {
        A[i] = i * 0.5f;
        B[i] = i * 0.25f;
    }

    struct timeval start, end;

    // Serial execution
    gettimeofday(&start, NULL);
    serial_matrix_add(A, B, C_serial, ROWS, COLS);
    gettimeofday(&end, NULL);
    printf("Serial time: %.3f ms\n", get_time_diff(start, end));

    // Vector execution
    gettimeofday(&start, NULL);
    vector_matrix_add(A, B, C_vector, ROWS, COLS);
    gettimeofday(&end, NULL);
    printf("Vector time: %.3f ms\n", get_time_diff(start, end));

    // Optional: verify correctness
    for (size_t i = 0; i < size; i++) {
        if (C_serial[i] != C_vector[i]) {
            printf("Mismatch at index %zu: %f != %f\n", i, C_serial[i], C_vector[i]);
            break;
        }
    }

    free(A);
    free(B);
    free(C_serial);
    free(C_vector);

    return 0;
}
```

**Expected behavior**
The simulation should complete without error and produce correct results for all supported VLEN values, including those greater than 512. Since RVV is designed to support scalable vector lengths, larger VLEN configurations should not crash the simulation.

**Host Operating System**
Ubuntu 22.04

**Host ISA**
X86

**Additional information**
Works fine for VLEN <= 512

erin-le, 2025-04-03:
@saul44203: Hello, would you be able to take a look at this issue? Thank you!

ntampouratzis, 2025-04-09:
Dear @saul44203 thank you for the response and your time. You are correct that it is working perfectly in SE mode (I have just tested), however, in FS mode I get segmentation with the provided ./configs/example/gem5_library/riscv-ubuntu-run.py script,  provided Ubuntu 24.04 and provided 6.6.33 Linux kernel.

To be noticed that doubling the VLEN (for example, from 256 to 512) the speedup is only ~15%. Is it a memory limitation?