gem5/gem5 #2145: RVV buggy when using IREE generated binaries

alhussein-jamil, 2025-04-04:
Okay, here is a carefully rewritten version of your bug report suitable for a gem5 GitHub issue, formatted in Markdown. Remember to replace the placeholder information (like the commit hash, OS, ISA, compiler version) before submitting.

```markdown
# Incorrect RISC-V Vector (RVV) execution result for matrix transpose via IREE

**Describe the bug**

When simulating a RISC-V binary generated by IREE (which executes a simple PyTorch matrix transpose model), gem5 produces incorrect results. The task is to transpose an 8x8 matrix of f32 values initialized to 1.0. The expected output is an 8x8 matrix of 1.0s. However, the simulation yields a matrix where only the top half contains 1.0s, and the bottom half contains 0.0s.

This suggests a potential issue in gem5's RISC-V Vector Extension (RVV) implementation, possibly related to the handling of the vector length multiplier (`lmul`) or specific vector instructions like `vslidedown`/`vslideup`. The simulation uses `VLEN=128` and the IREE compilation specifies `--riscv-v-fixed-length-vector-lmul-max=8`. Given the input data type is f32 (`SEW=32`), `VLEN=128` allows processing 4 elements per vector register (`VLMAX = VLEN / SEW = 128 / 32 = 4`). The incorrect result (only the first 4 rows seem correct) might indicate that operations involving `lmul > 1` (using multiple vector registers for a single logical vector operation) are not handled correctly.

The workflow involves:
1.  Defining a simple `transpose` operation in PyTorch.
2.  Exporting the model to MLIR using `torch.export`.
3.  Compiling the MLIR to a RISC-V executable module (`.vmfb`) using the IREE compiler, targeting RVV.
4.  Running this module using IREE's RISC-V runtime (`iree-run-module`) within a gem5 simulation (SE mode).

**PyTorch Model:**
```python
import torch

class Transpose(torch.nn.Module):
    def __init__(self):
        super().__init__()

    def forward(self, x):
        return x.transpose(0, 1)

# Example usage (not run in gem5, just for context)
# model = Transpose()
# input_tensor = torch.ones(8, 8, dtype=torch.float32) # Input tensor of ones
# output_tensor = model(input_tensor)
# print(output_tensor) # Should print an 8x8 tensor of ones
```

**IREE Compilation Command:**
```bash
# Assuming ${mlir_file} points to the exported MLIR representation
# Assuming ${binary_name} is transpose.vmfb
iree-compile "${mlir_file}" \
  --iree-hal-target-backends=llvm-cpu \
  --iree-llvmcpu-target-triple=riscv64-unknown-elf \
  --iree-llvmcpu-target-abi=lp64 \
  --iree-llvmcpu-target-cpu-features=+m,+a,+f,+zvl128b,+v \
  --riscv-v-fixed-length-vector-lmul-max=8 \
  -o "${binary_name}"
```

**Affects version**
`develop` branch (Commit: `PLEASE_ADD_COMMIT_HASH_HERE`)

**gem5 Modifications**
None.

**To Reproduce**

**Prerequisites:**
*   A working gem5 build for RISC-V.
*   A RISC-V toolchain (e.g., the one used for IREE compilation).
*   IREE compiled for RISC-V host tools (`iree-compile`) and target runtime (`iree-run-module`).
*   The compiled model artifact (`transpose.vmfb`) generated using the IREE command above from a suitable MLIR input.
*   The RISC-V `iree-run-module` binary.

**Steps:**

1.  **Compile gem5:**
    ```bash
    # From the gem5 root directory
    scons build/RISCV/gem5.opt -j<N> # Replace <N> with number of cores
    ```
[... 173 lines cut ...]
# Expected output content
[[1. 1. 1. 1. 1. 1. 1. 1.]
 [1. 1. 1. 1. 1. 1. 1. 1.]
 [1. 1. 1. 1. 1. 1. 1. 1.]
 [1. 1. 1. 1. 1. 1. 1. 1.]
 [1. 1. 1. 1. 1. 1. 1. 1.]
 [1. 1. 1. 1. 1. 1. 1. 1.]
 [1. 1. 1. 1. 1. 1. 1. 1.]
 [1. 1. 1. 1. 1. 1. 1. 1.]]
```

**Host Operating System**
`PLEASE_SPECIFY_HOST_OS_HERE (e.g., Ubuntu 22.04)`

**Host ISA**
`PLEASE_SPECIFY_HOST_ISA_HERE (e.g., x86_64)`

**Compiler used**
`PLEASE_SPECIFY_COMPILER_AND_VERSION_HERE (e.g., GCC 11.4.0 or Clang 15.0.7)`

**Additional information**
*   The issue seems tied to vector operations spanning more elements than fit in a single vector register (`VLEN/SEW = 128/32 = 4`), suggesting `lmul > 1` or related addressing/masking/slide operations might be incorrectly simulated.
*   The IREE compilation explicitly allows `lmul` up to 8. It would be helpful to know which specific vector instructions are generated by IREE for the `transpose` operation in this configuration.
*   The simulation uses `RiscvO3CPU`. It might be worth testing with `MinorCPU` or `AtomicSimpleCPU` (if RVV is supported) to see if the issue persists, which could help isolate whether it's in the core functional implementation or related to the O3 pipeline specifics.
```

erin-le, 2025-04-10:
@alhussein-jamil Hello, thanks for opening this issue. Based on your description of the bug, it's possible that this issue could be fixed by https://github.com/gem5/gem5/pull/1955, which has been merged into the `develop` branch, or by https://github.com/gem5/gem5/pull/1712, which is still open. Could you try using PR 1712 and the develop branch and see if they work for you?

alhussein-jamil, 2025-05-05:
@erin-le Hello, we were trying to use some other framework rather than IREE, I have already been using develop,  I tried #1712 patch, it does not seem to fix the problem.

saul44203, 2025-05-26:
Hi @alhussein-jamil, could you check if #2307 fixes this? If not, I would appreciate if you could upload a static binary, that way it's easier for me to reproduce it.

erin-le, 2025-06-04:
@alhussein-jamil: Hello, just a reminder about this. It'd be great if we could get https://github.com/gem5/gem5/pull/2307 into the gem5 v25.0 release. The staging branch for the release has already been created, but bug fixes are still being accepted onto the staging branch. The staging branch will be merged into stable on June 12th at the earliest, so it'd be helpful if you could take a look at it soon. Thank you!

closed 2025-07-31 by erin-le