gem5/gem5 #2148: Potential bug on big endian hosts for "far" atomics

powerjg, 2025-04-04:
The current implementation atomic functors does not convert the guest memory data into host endianness before operating on the data.

See #2143 for a fix for RISC-V.

There may be issues in other places. Here are some potential links:

### In Arm code

I'm hesitant to change this code with all of the templating.

* https://github.com/gem5/gem5/blob/develop/src/arch/arm/isa/insts/amo64.isa#L208
* https://github.com/gem5/gem5/blob/develop/src/arch/arm/isa/insts/amo64.isa#L329
* https://github.com/gem5/gem5/blob/develop/src/arch/arm/isa/insts/amo64.isa#L348
* https://github.com/gem5/gem5/blob/develop/src/arch/arm/isa/insts/amo64.isa#L409

### In GPU code

The GPU uses specialized classes for each operation. I think I can update them correctly. I'll push a patch soon.

See https://github.com/gem5/gem5/blob/develop/src/base/amo.hh#L129

powerjg, 2025-04-04:
FYI: @abmerop @giactra 

I feel comfortable updating the GPU code which uses the more generic operations, but I don't feel comfortable updating the Arm code right now.

giactra, 2025-04-04:
Thanks Jason, I will look into it

mattsinc, 2025-04-05:
> FYI: [@abmerop](https://github.com/abmerop) [@giactra](https://github.com/giactra)
> 
> I feel comfortable updating the GPU code which uses the more generic operations, but I don't feel comfortable updating the Arm code right now.

Thanks @powerjg , I can also look at this once you push something