gem5/gem5 #2159: Ruby protocol MOESI_CMP_token broken after #1453

powerjg, 2025-04-08:
#1453 made a change where the `RubySystem` pointer has to be explicitly passed around. This does not happen for the `PersistentTable` object that is used in MOESI_CMP_token.

You can reproduce this issue with the following:

```sh
 build/ALL/gem5.opt configs/example/ruby_random_test.py -n 4 --protocol=MOESI_CMP_token
```

The expected behavior is that the test runs to completion. The actual result is an assert failure.

Unfortunately, with a quick skim I can't find a simple fix. We may want to simply delete this protocol.

I don't want to call this a "bug" per-se. It *is* a bug, but it's in code that we have not been supporting for probably decades.

powerjg, 2025-04-08:
@abmerop do you happen to see an easy way to fix this? If not, I advocate for dropping this protocol.

Here's the code that needs to change: https://github.com/gem5/gem5/blob/develop/src/mem/ruby/structures/PersistentTable.hh#L52

abmerop, 2025-04-08:
I thought I had handled this case. I think something like this would work:

1) Add `"PersistentTable"` to the list here https://github.com/gem5/gem5/blob/develop/src/mem/slicc/symbols/StateMachine.py#L836 (or below if it is a _ptr)
2) Define `setRubySystem` in `PersistentTable`
3) Call `setRubySystem` for each `NetDest` in `PersistentTableEntry` when it is created.

Thinking about it more, I believe I couldn't figure out why `PersistentTableEntry` is `static`: https://github.com/gem5/gem5/blob/develop/src/mem/ruby/structures/PersistentTable.cc#L52 and the protocol looked unused so I may not have finished a commit for it.

powerjg, 2025-04-08:
That will work for `PersistentTable`, but since the entries are in a map it's going to take a bit deeper change to fix. We'll have to `emplace` or provide a default implementation for the creator for the map as well.

This is low priority, I think.