gem5/gem5 #2164: Weird deadlocks with -n > 18 in ruby_random_test.py.

Mayyygic, 2025-04-09:
**Describe the bug**
I've been using ruby_random_test.py to test my own protocol. Everything works fine with -n <= 16 and --maxloads=50,000,000, but when I increase -n to 32, it hits a deadlock.

At first, I suspected a bug in my protocol and enabled ProtocolTrace and RubySlicc, but found nothing wrong. Then, with RubyQueue, I noticed one processor/node becomes idle right after initialization and is marked as deadlocked after 50,000 Cycles— but it hasn't done anything yet.

This seems to be a bug in RubyTester. I confirmed the same issue using build/X86/gem5.opt — the deadlock appears when -n is greater than 18.

**Affects version**
My gem5 version is 24.1.0.1

**gem5 Modifications**
No changes except my own protocol sm files.

**To Reproduce**
1. Compile gem5 with command: scons -j$(nproc) build/X86/gem5.opt 
2. Execute the simulation: build/X86/gem5.opt --debug-flags=RubyQueue --debug-file=trace_q.log configs/example/ruby_random_test.py -n19 --maxloads=100

**Terminal Output**

```shell
src/cpu/testers/rubytest/RubyTester.cc:271: panic: Deadlock detected: current_time: 50001 last_progress_time: 0 difference:  50001 processor: 2
```

**Host Operating System**
Ubuntu 22.04

**Additional information**
If you search for l1_cntrl2 in the trace_q.log file, you will find that it has only 7 entries related to initialization.

Mayyygic, 2025-04-10:
I think the deadlock detection logic in RubyTester should focus only on processors that are actively participating in the simulation. If a processor hasn't performed any action or check, it's effectively idle and shouldn't be considered in the deadlock check.

From an implementation perspective, we could introduce a std::vector<bool> (or similar) in RubyTester to track active processors. When a packet is successfully sent to a processor, we set the corresponding flag. When a tester callback is triggered (i.e., a response is received), we clear it. Then, during each wakeup, the deadlock check should only evaluate processors marked as active.

powerjg, 2025-04-10:
That makes sense. However, why do we have processors that are not "actively participating" shouldn't all of the testers be sending requests to all of the caches?

Mayyygic, 2025-04-11:
I've fixed the problem by updating the deadlock detection logic in RubyTester.

Now, whenever a Check successfully sends a request, it's recorded under the corresponding processor. Once the request is completed, it's removed.

During each wake-up cycle, RubyTester now checks whether each processor currently has any active requests. If it does, it iterates over those requests to determine if any of them are deadlocked.

If this solution seems reasonable, I will proceed to create a pull request for your review.

Mayyygic, 2025-04-29:
@erin-le Sorry for my late reply, I will raise a PR in a few days!

closed 2025-05-08 by erin-le