gem5/gem5 #2185: Make gem5 a set of shared objects

hchsiao, 2025-04-15:
It was mentioned in PR #2172 that we may turn gem5 into several shared objects (e.g., o3cpu.so, arm_isa.so, etc.) plus a small core executable to make gem5 more modularized. My (probably too optimistic) interpretation of this task is that we can done this gradually by decoupling feature groups one-by-one, such as

1. **(current)** The monolithic gem5 binary (featureX, featureY, featureZ, ...) + no shared object
2. gem5 binary (featureX, featureY, featureZ, ...) + trivial.so
3. gem5 binary (featureY, featureZ, ...) + featureX.so
4. gem5 binary (featureZ, ...) + featureX.so + featureY.so
5. ... (repeat until nothing can be decoupled)

Theoretically, this has nothing to do with changing the build system. But I believe Bazel can be particularly helpful along the way in decoupling featureXYZ (from 2. to 3. and 4.). However, to move from 1. to 2., a new infrastructure needs to be devised to extend gem5 with shared objects. Can someone shed some light for me on this specifically?

hchsiao, 2025-04-24:
Hi @powerjg, could you provide some insight on how this task should be initiated? I'll be more than happy to work on this one, or perhaps I can try creating a prototype and we can make more discussion based on that?

rogerchang23424, 2025-04-28:
Hi maintainers,

Could you reply it?

powerjg, 2025-04-28:
Sorry for the slow reply... I was backpacking last week.

> Theoretically, this has nothing to do with changing the build system. But I believe Bazel can be particularly helpful along the way in decoupling featureXYZ (from 2. to 3. and 4.).

Fully agreed! Bazel vs making gem5 more modular are orthogonal. However, I think that both making gem5 more modular is (1) going to make a lot of the Bazel things easier and (2) having Bazel will make proving that the decoupling work easier.

> My (probably too optimistic) interpretation of this task is that we can done this gradually by decoupling feature groups one-by-one, such as

I think your plan is great. 

One thing that's missing is that we are going to have to move some files around during this process. Moving the files around is going to be disruptive (but probably worth it in the long run). We'll have to tread carefully while doing it.

In fact, you can generally say that this change is going to be highly disruptive. So, we need to make sure that it's *worth* the pain. IMO, if we get to a point that we can distribute a (small) gem5 binary, people can download pre-built plugins, and they can develop their own plugins without having to use the entire gem5 source tree, then it's worth a significant amount of pain. On the other hand, if the goal is to just "use the cool build system on the block" or "make gem5 compile slightly faster" then it's definitely *not* worth it.

> **Side note** many years ago, someone took each directory in the gem5 source and made each directory compile into its own library file (statically, not dynamically). The idea was that it would save compile time by doing partial linking before the big linking step and the end. It took a non-insignificant amount of effort to make even that "simple" modularization change. However, within a year or two, we had to revert all of the work. I don't remember exactly why. It could have been because some compiler wasn't behaving in the way expected or scons didn't allow the partial linking feature in the way we needed it to work. Either way, the work was wasted and ended up actually causes us more pain than it created benefits.

I bring up that anecdote just to make sure that we aren't repeating mistakes of the past. If we are going to make a (highly) disruptive change for our many 1000s of users, we need to make sure that it's worth (1) all of those people changing their workflows and (2) **that we plan a way to document and teach everyone about the changes**

hchsiao, 2025-04-29:
Thanks, @powerjg. Your input was very constructive especially it has addressed many of my concerns about user impact and the project's goals/non-goals. I'll start working on this.

powerjg, 2025-04-29:
Sounds great! Including @BobbyRBruce in the conversation as well as this has been on his todo list for years 🙂

BobbyRBruce, 2025-04-29:
Yes! This is something I really support.

We're going to discuss improving the modularity of gem5 in the next developers meeting. Anyone is free to join so please feel free to if you wish to join this discussion.

Right now I'm starting to work out how we can improve our build system to make this sort of thing possible. I'll hopefully have something to talk about come next week on the topic.

Harshil2107, 2025-04-29:
Hi,

I have added a topic on the modularity of gem5 to this month’s developers meeting agenda. It will be an open discussion, and everyone is welcome to join.

You can find the discussion post here: [gem5 Dev Meeting](https://github.com/orgs/gem5/discussions/2224)

powerjg, 2025-04-29:
@hchsiao I assume that due to time zones the dev meeting is going to be difficult for you to attend (correct me if I'm wrong!).

We need to find a better schedule that will enable people from all over the world to attend the dev meetings. This *will* happen, but probably not until June or July.

hchsiao, 2025-04-30:
Yes, me and my team are based in Taiwan. As we're still brainstorming technical aspects of the task, I don't have much to update for the upcoming meeting yet. However, shifting the dev meeting occasionally would be great for sure!

BobbyRBruce, 2025-04-30:
> Yes, me and my team are based in Taiwan. As we're still brainstorming technical aspects of the task, I don't have much to update for the upcoming meeting yet. However, shifting the dev meeting occasionally would be great for sure!

The meetings are recorded and put on our youtube channel, www.youtube.com/@gem5, so can catch up on anything said there if unable to attend. We also publish AI generated meeting notes in the event's GitHub Discussions page (https://github.com/orgs/gem5/discussions/2224)

hchsiao, 2025-05-09:
I've been tinkering with the codebase these days and have some initial findings on modularization. Would you be able to create a design document if there's a preferred place to host the doc? Or I can also draft my findings first using google doc.

powerjg, 2025-05-12:
A sharable google doc is great. Alternatively, something in this issue or on github discussions works.

One thing to keep in mind from the dev meeting is that the broader community is hesitant to accept big changes (like this or like changing the build system) unless the community is strongly convinced of big positive change.

BobbyRBruce, 2025-05-12:
For those interested in referencing what was discussed during the meeting, a recording of the event can be found here: https://youtu.be/LQJhJgbL4q4/. The AI meeting notes and general discussions regarding the meeting and its contents can be found on the event's GitHub Discussions page: https://github.com/orgs/gem5/discussions/2224

hchsiao, 2025-05-13:
The change is indeed significant. However, I believe the task of modularizing gem5 can still be broken down into smaller subtasks. For example, coupled code needs to be decoupled before modularization can be achieved. This holds true regardless of how the API will be implemented or whether dlopen or the OS loader will be used for loading the `.so` files. The boundaries for code separation cannot be effectively drawn until the components are made reasonably self-contained through refactoring. In my opinion, the risk associated with this effort toward "partitioned" code should be minimal, while the payoff is high, assuming modularization is the desired outcome. Here I want to share some of my findings on the codebase "partitioning" journey.

Consider a partitioning approach that divides the codebase into a "core simulator" (containing common code essential for all gem5 users) and "the rest." It's evident that the "core simulator" is a dependency of "the rest," but not vice versa. Speaking of common code, my initial attempt involves incorporating `src/sim/init.cc`, as it implements the `EmbeddedPyBind` functionalities, and `src/python/pybind11/*.cc` for the `_m5` components.

Bazel now comes into play! We can leverage the prototype from #2172 to `bazel query` the build database (We are only using bazel to obtain insights on the dependency graph and make changes accordingly, so no need to merge the PR). The following query basically translates to: "Hey Bazel, what else should I include for the core simulator in addition to these given files?".
```
bazel cquery --noimplicit_deps "deps(set(//src/sim:init //src/python/pybind11:core //src/python/pybind11:event //src/python/pybind11:stats //src/python/pybind11:debug))"
```

The output (with some pruning):
```
src/python/pybind11/pybind.{cc,hh}
src/sim/init.{cc,hh}
src/cpu/probes/pc_count_pair.{cc,hh}
src/base/statistics.{cc,hh}
src/base/socket.{cc,hh}
src/base/debug.{cc,hh}
src/sim/eventq.{cc,hh}
src/sim/core.{cc,hh}
src/base/types.{cc,hh}
src/base/output.{cc,hh}
src/base/loader/elf_object.{cc,hh}
src/sim/stat_control.{cc,hh}
src/sim/sim_exit.{cc,hh}
src/python/pybind_init.{cc,hh}
src/base/cprintf.{cc,hh}
src/base/compiler.{cc,hh}
src/sim/sim_events.{cc,hh}
src/sim/simulate.{cc,hh}
src/sim/cur_tick.{cc,hh}
src/base/stats/text.{cc,hh}
src/base/random.{cc,hh}
src/sim/serialize.{cc,hh}
src/base/inet.{cc,hh}
src/base/temperature.{cc,hh}
src/base/trace.{cc,hh}
src/sim/sim_object.{cc,hh}
src/sim/debug.{cc,hh}
src/base/addr_range.{cc,hh}
src/base/logging.{cc,hh}
src/sim/stat_register.{cc,hh}
src/sim/drain.{cc,hh}
src/base/str.{cc,hh}
src/base/match.{cc,hh}
src/base/atomicio.{cc,hh}
src/base/callback.{cc,hh}
src/sim/init_signals.{cc,hh}
src/base/time.{cc,hh}
src/sim/system.{cc,hh}
src/sim/stats.{cc,hh}
src/base/cprintf_formats.{cc,hh}
src/base/stats/group.{cc,hh}
src/base/stats/output.{cc,hh}
src/base/uncontended_mutex.{cc,hh}
src/sim/kernel_workload.{cc,hh}
src/base/pollevent.{cc,hh}
src/base/named.{cc,hh}
src/cpu/pc_event.{cc,hh}
src/sim/async.{cc,hh}
src/base/cast.{cc,hh}
src/base/loader/object_file.{cc,hh}
src/sim/port.{cc,hh}
src/base/loader/symtab.{cc,hh}
src/sim/serialize_handlers.{cc,hh}
src/cpu/smt.{cc,hh}
src/dev/net/etherpkt.{cc,hh}
src/base/stats/units.{cc,hh}
src/base/type_traits.{cc,hh}
src/base/stats/storage.{cc,hh}
src/sim/byteswap.{cc,hh}
[... 58 lines cut ...]

Bazel then shows:
```
//src/python/pybind11:core
//src/base:inet
//src/dev/net:etherpkt.hh
```

In this case, we may revisit `src/python/pybind11/core.cc` and see if it really needs to depend on `src/base/inet.*`.

Another example illustrating sim_object dependency:
```
bazel cquery --noimplicit_deps "somepath(//src/python/pybind11:all, //src/sim/power:ThermalReference_param_hh)"
```

And the path is valid:
```
//src/python/pybind11:debug (3723722)
//src/sim:debug (3723722)
//src/generated/params:System (3723722)
//src/generated/params:System_py (3723722)
//src/generated/params:ThermalModel_py (3723722)
//src/sim/power:ThermalReference_param_hh (3723722)
```
Which seems to be worth decoupling as well.