gem5/gem5 #2188: Reverting PR 2174

erin-le, 2025-04-16:
This issue has been opened to record the reason for reverting PR 2174. PR 2174 was reverted due to the following failures with the daily tests:

For `testlib-long-tests (fs)`,  `realview64-o3-dual-ruby-ARM_X86-x86_64-opt` and its MatchRegex failed. The following is printed in simerr:
```
src/mem/ruby/system/RubyPort.cc:283: warn: Cache maintenance operations are not supported in Ruby.
gem5.opt: src/cpu/o3/lsq.cc:1164: virtual bool gem5::o3::LSQ::SingleDataRequest::recvTimingResp(gem5::PacketPtr): Assertion `_numOutstandingPackets == 1' failed.
Program aborted at tick 4489520001
--- BEGIN LIBC BACKTRACE ---
```
For `testlib-long-tests (x86_boot_tests)`, `timing-cpu_1-cores_mesi_two_level_DualChannelDDR3_1600_systemd_x86-boot-test-ALL-x86_64-opt` failed. The following is printed in simerr:

```
src/arch/generic/debugfaults.hh:145: warn: MOVNTDQ: Ignoring non-temporal hint, modeling as cacheable!
build/ALL/mem/ruby/protocol/MESI_Two_Level/Directory_Transitions.cc:268: panic: Invalid transition
board.cache_hierarchy.ruby_system.directory_controllers0 time: 15159272260 addr: 0xba174e00 event: CleanReplacement state: M_DRD
Memory Usage: 4323588 KBytes
Program aborted at tick 5048037662580
--- BEGIN LIBC BACKTRACE ---
```

saul44203, 2025-04-23:
Hi @erin-le, thanks, will take a look at this. I'll post any updates here.

saul44203, 2025-04-23:
@erin-le does any of those tests software prefetch instructions?

powerjg, 2025-04-25:
> Maybe a silly question, but does any of those tests have software prefetch instructions?

Most likely, yes. They are full Linux boot tests with a full usermode system (i.e., systemd).

erin-le, 2025-08-13:
Closing as PR 2311 has been merged.

closed 2025-08-13 by erin-le