gem5/gem5 #2211: Disabling both RAS and indirectHashTargets in branch predictor causes risc-v O3 simulation failure

faridsamandi, 2025-04-25:
**Describe the bug**
When I disable return address stack (RAS), and hashing path history targets for the indirect branch predictor, both together, the O3 simulation faults and exits very early into the simulation with this error:
`src/mem/xbar.cc:368: fatal: Unable to find destination for [0xffffffffffffffc0:0] on board.cache_hierarchy.l2buses`

**Affects version**
branch: stable (v24.1.0.3), commit https://github.com/gem5/gem5/commit/b9da2bfe1e21f6efa2730edd4e80e58c5d965458

**gem5 Modifications**
Modify branch predictor parameters file (`src/cpu/pred/BranchPredictor.py`): set `indirectHashTargets` to False, and set `ras` to NULL.
Change cpu type to O3 in `configs/example/gem5_library/riscv-fs.py`

**To Reproduce**
1. Compile gem5 with command `scons build/RISCV/gem5.opt`
**Interestingly**, this problem only happens after rebuilding gem5 (i.e., if you build gem5 with default parameters and then change them, the simulation doesn't fault.) I had the impression that changing parameter files (e.g., "src/cpu/pred/BranchPredictor.py") doesn't require rebuilding gem5. Is my understanding incorrect?
3. Execute the simulation with `./build/RISCV/gem5.opt ./configs/example/gem5_library/riscv-fs.py`

**Terminal Output**
```shell
$ ./build/RISCV/gem5.opt --debug-flags=PortTrace ./configs/example/gem5_library/riscv-fs.py
gem5 Simulator System.  https://www.gem5.org
gem5 is copyrighted software; use the --copyright option for details.

gem5 version 24.1.0.3
gem5 compiled Apr 25 2025 03:34:50
gem5 started Apr 25 2025 03:37:20
gem5 executing on killians, pid 1454762
command line: ./build/RISCV/gem5.opt --debug-flags=PortTrace ./configs/example/gem5_library/riscv-fs.py

info: Using default config
Beginning simulation!
Global frequency set at 1000000000000 ticks per second
      0: board.platform.rtc: Real-time clock set to Sun Jan  1 00:00:00 2012
board.platform.terminal: Listening for connections on port 3456
src/arch/riscv/isa.cc:279: info: RVV enabled, VLEN = 256 bits, ELEN = 64 bits
src/base/statistics.hh:279: warn: One of the stats is a legacy stat. Legacy stat is a stat that does not belong to any statistics::Group. Legacy stat is deprecated.
board.remote_gdb: Listening for connections on port 7000
src/arch/riscv/linux/fs_workload.cc:166: info: Bootloader is not specified.
src/arch/riscv/linux/fs_workload.cc:180: info: Loaded kernel '/var/services/home/.cache/gem5/riscv-bootloader-vmlinux-5.10' at 0x0
src/arch/riscv/linux/fs_workload.cc:199: info: Loaded DTB 'm5out/device.dtb' at 0x87e00000
src/sim/simulate.cc:199: info: Entering event queue @ 0.  Starting simulation...
src/mem/xbar.cc:368: fatal: Unable to find destination for [0xffffffffffffffc0:0] on board.cache_hierarchy.l2buses
Port trace of the Packet (
[Destination] board.cache_hierarchy.l2buses.cpu_side_port[1]
[Source] board.cache_hierarchy.l1d-cache-0.mem_side_port addr=0xffffffffffffffc0
)
Memory Usage: 8561372 KBytes
```

**Host Operating System**
Ubuntu 22.04

**Host ISA**
X86

powerjg, 2025-05-01:
I don't have any specific ideas on the bug, maybe @dhschall can help as he's been working a lot with the branch predictors. I do have one small comment, though (to answer your question).

> I had the impression that changing parameter files (e.g., "src/cpu/pred/BranchPredictor.py") doesn't require rebuilding gem5. Is my understanding incorrect?

You shouldn't ever modify parameters in what I like to call "SimObject description files." The idea is that the SimObject description file *declares the parameters*. Then, you can change the parameters in your *configuration* file or *run script*. 

**If you make any changes in `src/`** then you have to recompile.

dhschall, 2025-05-01:
Hi @faridsamandi, I've tried to reproduce the error but I could not. Maybe I used something different configuration.
However, I realized that the `riscv-fs.py` uses the `BaseTimingSimpleCPU`, which by default has no branch predictor at all. Did you also make some changes there?

If I configure the `riscv-fs` config file with the parameters you described, it seems to work for me, and Linux boots.
Can you try if the following works for you as well?

```Python
# Setup a single-core Processor.
processor = SimpleProcessor(
    cpu_type=CPUTypes.TIMING, isa=ISA.RISCV, num_cores=1
)

from m5.objects import (
    NULL,
    TournamentBP,
)

cpu = processor.cores[0].core
cpu.branchPred = TournamentBP()
cpu.branchPred.ras=NULL()
cpu.branchPred.indirectBranchPred.indirectHashTargets = False
```

faridsamandi, 2025-05-01:
Hi @dhschall , the problem only occurs for O3 cpu. The configuration you pasted above works fine for me as is. But if I change the `cpu_type` to `O3`, the simulation faults with the same error message.

dhschall, 2025-05-02:
Thank you for the clarification! I could reproduce the error if I set the `cpu_type` to `O3`.
I also found that the bug is caused when I set `cpu.branchPred.ras=NULL()`. The indirect predictor doesn't cause any issue.

I traced the instructions and I the reason is that without the RAS, the branch predictor continues speculating past the function ends and decodes instructions that will never be committed. These instructions seem to generate an invalid address that causes the fault. However, these instructions are miss-speculated and would have never been committed.

Note that such a situation is not unusual. E.g., a modern processor would also miss the return if it's not in the BTB (can be simulated by setting `cpu.branchPred.requiresBTBHit = True`). (I've tried that, but it seems that in this case we are lucky and the load is squashed before it could have issued the memory request.)

To fix the issue, we could add a check in the LSQ Unit if the address is outside of the physical range, as is done in the fetch unit for instructions. See the attached patch. Does it work for you?

[lsq_diff.txt](https://github.com/user-attachments/files/20010132/lsq_diff.txt)

faridsamandi, 2025-05-03:
@dhschall , thanks for tracking down the issue. 
Your solution doesn't work. The simulation stalls waiting for a load instruction in the bootloader to complete. I haven't got the chance to dig further into it though.