gem5/gem5 #2253: Potential Bug: Locked address mask in cpu and memory is inconsistent

rogerchang23424, 2025-05-06:
**Describe the bug**
Locked address in ARM
https://github.com/gem5/gem5/blob/41b9c344d7603e2f4dad43d3248e6b9a23ac75cb/src/arch/arm/isa.cc#L1727-L1738

Locked address in RISCV
https://github.com/gem5/gem5/blob/41b9c344d7603e2f4dad43d3248e6b9a23ac75cb/src/arch/riscv/isa.cc#L1012-L1032

Locked address in AbstractMemory
https://github.com/gem5/gem5/blob/41b9c344d7603e2f4dad43d3248e6b9a23ac75cb/src/mem/abstract_mem.hh#L67-L101

**Expected behavior**
Is the Locked address mask need to consistent?

**Host Operating System**
Ubuntu 22.04

**Host ISA**
X86

powerjg, 2025-05-08:
Thanks for pointing this out... could you be more specific on what is inconsistent and what you think should change?

Can you provide an example where gem5 misbehaves?

rogerchang23424, 2025-05-09:
Updated, just question about if the reservation size need to consistent.

powerjg, 2025-05-12:
What effect would this difference have? Maybe in atomic mode this could be a problem if the lock was pushed all the way down to the memory? Maybe if the CPUs were directly connected to an abstract memory without caches?

I think that with a CPU model + caches the CPU model and/or the caches would be the thing that handles the lock. I don't think the lock `MemCmd` gets sent all the way to the abstract memory. I could be mistaken though.

This is a good catch, but I am very hesitant to change anything unless we can come up with a case where gem5 behaves incorrectly.