gem5/gem5 #2265: SE results are dependent on command pathname

brian-schwedock, 2025-05-10:
**Describe the bug**
When running in SE mode, different pathnames for the same command file provide different simulation results. Running with the same pathname provides identical results.

**Affects version**
v24.1.0.3

**gem5 Modifications**
None

**To Reproduce**
1. Compile: `scons build/ARM/gem5.opt`
2. Run gem5: `./build/ARM/gem5.opt configs/example/arm/starter_se.py tests/test-progs/hello/bin/arm/linux/hello`
3. Copy the stats file: `cp m5out/stats.txt stats-1.txt`
4. Run the same experiment with a different pathname for the same command: `./build/ARM/gem5.opt configs/example/arm/starter_se.py tests/../tests/test-progs/hello/bin/arm/linux/hello`
5. Copy the stats file: `cp m5out/stats.txt stats-2.txt`
6. Compare results: `diff stats-1.txt stats-2.txt`

**Terminal Output**
```
...
< system.cpu_cluster.cpus.numCycles                5841                       # Number of cpu cycles simulated (Cycle)
< system.cpu_cluster.cpus.cpi                  1.160310                       # CPI: cycles per instruction (core level) ((Cycle/Count))
< system.cpu_cluster.cpus.ipc                  0.861839                       # IPC: instructions per cycle (core level) ((Count/Cycle))
---
> system.cpu_cluster.cpus.numCycles                5948                       # Number of cpu cycles simulated (Cycle)
> system.cpu_cluster.cpus.cpi                  1.158325                       # CPI: cycles per instruction (core level) ((Cycle/Count))
> system.cpu_cluster.cpus.ipc                  0.863315                       # IPC: instructions per cycle (core level) ((Count/Cycle))
...
```

**Expected behavior**
I expect results to be identical, as is the case when providing the identical pathname.

**Host Operating System**
Ubuntu 22.04

**Host ISA**
X86

**Compiler used**
GCC 11.4.0

powerjg, 2025-05-12:
This is a known behavior.

Because SE mode uses the *host* information, there is no way to provide perfectly deterministic results. Basically, part of the cstdlib will read the path to the executable. In the two cases, this path is a different length, so you get (slightly) different code path executions.

I'm a little surprised it's not printed in `hello`, but we have warnings like this: https://github.com/gem5/gem5/blob/develop/src/sim/syscall_emul.hh#L1121

If you need deterministic results, you have to use full system mode (and *not* use KVM).

brian-schwedock, 2025-05-13:
It looks like the code uses the absolute path of the binary though, and in my example both executions use the same absolute path. https://github.com/gem5/gem5/blob/develop/src/sim/syscall_emul.hh#L1105

erin-le, 2025-05-15:
@brian-schwedock Hello, thanks for opening this issue. Can you clarify if the intent of this issue to report a bug? If so, do you have a fix for it on your end?

brian-schwedock, 2025-05-15:
The intent is to report a bug. I do not have a fix.