gem5/gem5 #2300: Broken link in gem5-resources-website

santimc, 2025-05-23:
In [riscv-linux-6.5.5-kernel](https://resources.gem5.org/resources/riscv-linux-6.5.5-kernel?database=gem5-resources&version=1.0.0) the link to repository is https://github.com/gem5/gem5-resources/tree/stable/src/riscv-opensbi and it's broken, should be https://github.com/gem5/gem5-resources/tree/stable/src/riscv-fs/riscv-opensbi

I'm opening the issue here because it's not allowed in the [gem5-resources-website repo](https://github.com/gem5/gem5-resources-website), I hope it's ok.

I cant also send a pull request because this specific page looks like it's beeing pulled from a external mongo db and not the resources.json

powerjg, 2025-05-27:
Thanks for letting us know. We'll get this fixed!

Harshil2107, 2025-07-14:
I have updated the database. The link on the website should now point to the correct repository.