gem5/gem5 #2306: IndexError: _Map_base::at

12345678-5816, 2025-05-26:
**Based on Gem5 platform，I have done any extension of DRAM model for LPDDR5X@9600Gbps. And the changes are mainly in ext/dramsys/DRAMSys. 
      While I  do any test for the new DRAM model based on configs/example/dramsys.py，I found there is any errors  which are shown as follows：**

/************************************************************************************************
command line: ./build/***/gem5.debug ./configs/example/dramsys.py

Global frequency set at 1000000000000 ticks per second
warn: failed to generate dot output from m5out/config.dot

\u25a0 \u25a0 \u25a0  DRAMSys5.0, Copyright (c) 2023
\u25a0 \u25a0 \u25a0  RPTU Kaiserslautern-Landau,
\u25a0 \u25a0 \u25a0  Fraunhofer IESE

IndexError: _Map_base::at

At:
  src/python/m5/SimObject.py(1262): getCCObject
  src/python/m5/SimObject.py(1285): getValue
  src/python/m5/params.py(262): getValue
  src/python/m5/SimObject.py(1217): getCCParams
  src/python/m5/SimObject.py(1281): createCCObject
  src/python/m5/simulate.py(135): instantiate
  ./configs/example/dramsys.py(62): <module>
  src/python/m5/main.py(669): main

/*************************************************************************************************

      Have you met the upper errors and could you please share any workaround to me？

powerjg, 2025-05-27:
Can you provide us with the necessary changes/configs/inputs to reproduce this problem?

12345678-5816, 2025-06-06:
> Can you provide us with the necessary changes/configs/inputs to reproduce this problem
The Changes I have done just under the directory of "ext/dramsys/DRAMSys" in the platform of GEM5 and i only added any files which related with LPDDR5T.

erin-le, 2025-07-03:
@12345678-5816: Hello, could you provide us with any files that you've added or changed that will help us reproduce the issue? The files you provide don't have to be the exact same as what you have, and can be a more minimal example, as long as they're still able to reproduce the issue.

Also, DRAMSys integration with gem5 has been updated so gem5 now uses DRAMSys v5.1 (see https://github.com/gem5/gem5/pull/2093) as of gem5 v25.0. You may find it helpful to try using the latest version of gem5. Thanks for opening this issue!

closed 2025-08-01 by github-actions[bot]