gem5/gem5 #2332: Suggest documentation update for building on low-spec systems ( -j$(nproc --ignore=2) and nice)

Sohail103, 2025-06-01:
Hi! I’d like to suggest a small addition to the documentation to help users building gem5 on lower-end systems.

When building with:

`scons build/X86/gem5.opt -j$(nproc)`

on a machine with limited RAM and swap, my system became unresponsive and the build eventually failed with the following error:

```
collect2: fatal error: ld terminated with signal 9 [Killed]
compilation terminated.
scons: *** [build/X86/gem5.opt] Error 1
scons: building terminated because of errors.
```

I believe this is due to memory exhaustion: the linker (`ld`) was killed by the system (likely by the OOM killer). This can happen even if CPU cores are available, especially when RAM is limited or swap is small/slow.

Using the following command worked for me:

`nice -n 19 scons build/X86/gem5.opt -j$(nproc --ignore=2)`

This reduces the priority of the build and reserves a couple of cores, helping the system stay responsive and preventing crashes.

I think a short note like this in the `README.md` or the build guide could be helpful for users on lower-spec machines.

This is my first issue on gem5, and I’d love to open a small PR for this if it sounds good!
Thanks again for all your work maintaining gem5!

powerjg, 2025-06-02:
Out of curiosity, are you modifying gem5, or just obtaining a binary to use? If the latter, I would suggest using a pre-built docker image instead of building it yourself.

I agree that we need to improve the documentation around this (it's a common error given how many posts there have been about it), but I think we'll change the documentation to say something like "building gem5 is a compute and memory intensive process that requires a high-end machine."

What do you think of this?

Sohail103, 2025-06-03:
I'm very new to the simulator, I heard about gem5 pretty recently and was just following the documentation to install it. I’m not modifying gem5 yet so I was mainly trying to get it up and running but I do aim to contribute in the future so I’m glad I got some experience with the build process.

I completely understand that the build is compute and memory intensive and I think updating the documentation to reflect that would definitely help set expectations. That said, I think it would be even more helpful if there was a note about building with `nice` or a similar way to reduce system impact, especially for users on older or shared machines. I initially tried building with 4 threads and then with 2 threads using scons, but both attempts failed. Eventually, I was able to build gem5 on an 11th-gen i3 laptop with 8 GB RAM using just one thread and low priority. It took around 3 hours, but the system stayed responsive throughout.

Suggesting Docker is great for users who only need pre-built binaries, but it might feel a bit dismissive to say a high-end machine is required, especially since Docker doesn’t make source-level experimentation particularly easy. From what I understand, anyone wanting to modify or add features to gem5 would generally need access to the entire source tree, including build scripts, source files, and headers, to compile and test changes. I imagine many learners and contributors start by building from source to eventually understand and tinker with the code.

Also, I wanted to reach out to the community and ask this on Slack, but the invite link in the README seems to be broken. Just thought I’d mention that in case it needs updating.

erin-le, 2025-06-26:
@Sohail103: Hello, I've opened https://github.com/gem5/website/pull/172 to update the building documentation and https://github.com/gem5/website/pull/176 to update the Slack link. Thanks for reporting these issues!

closed 2025-06-26 by erin-le