gem5/gem5 #2410: gem5-bridge readfile is not working

ntampouratzis, 2025-06-28:
**Describe the bug**
When executing riscv-ubuntu-run.py with a custom boot script passed via the `--script` argument (e.g., `--script=my_script.rcS`) in gem5 v25.0, the simulation fails during script execution inside the guest. Specifically, the `gem5-bridge readfile` mechanism fails to properly copy the script to `/tmp/script`. As a result, `/home/gem5/after_boot.sh` tries to execute `/tmp/script` and returns the error:

`/home/gem5/after_boot.sh: line 51: /tmp/script: cannot execute binary file: Exec format error`

This issue was not present in gem5 v24.x, where the same setup successfully executed the script.

**Affects version**
gem5 v25.0 (latest release as of June 2025)

**gem5 Modifications**
Added `no_systemd=True` to the board instantiation in the last line of `gem5/src/python/gem5/components/boards/riscv_board.py` to speed up boot.

Modified `riscv-ubuntu-run.py` to include the following logic:
```
import os, argparse

parser = argparse.ArgumentParser(
    description="A script which uses the RISCV in FS mode."
)
parser.add_argument("--script", type=str, default="", help="Linux bootscript")
args = parser.parse_args()

...
board.readfile = args.script
```

**To Reproduce**
1. Build gem5 for RISCV:

`scons build/RISCV/gem5.opt -j$(nproc)`

2. Create a simple script file my_script.rcS:
```
#!/bin/sh
echo "Hello from inside the simulated system!"
```

3. Run the simulation using:
`./build/RISCV/gem5.opt configs/example/gem5_library/riscv-ubuntu-run.py --script=my_script.rcS`

4. Observe that the boot completes, but the guest system reports:
```
[    0.129686] VFS: Mounted root (ext4 filesystem) on device 254:1.
[    0.130139] devtmpfs: mounted
[    0.130782] Freeing unused kernel image (initmem) memory: 2200K
[    0.130841] Run /sbin/init as init process
Kernel booted, In gem5 init...
Dropping to shell as gem5 user...
-bash: cannot set terminal process group (-1): Inappropriate ioctl for device
-bash: no job control in this shell
In after_boot.sh...
Interactive mode: false
Starting gem5 init... trying to read run script file via readfile.
Running script from gem5-bridge stored in /tmp/script
/home/gem5/after_boot.sh: line 51: /tmp/script: cannot execute binary file: Exec format error
Done running script from gem5-bridge, exiting.
```

**Expected behavior**
The file `my_script.rcS` should be copied into the simulated system via readfile and executed by `/home/gem5/after_boot.sh`. This behavior was functional in gem5 v24.x.

**Host Operating System**
Ubuntu 22.04,

**Host ISA**
X86

powerjg, 2025-06-30:
@Harshil2107 can you take a look when you get back in a couple of weeks?

@ntampouratzis as a quick test... can you use interactive mode and try to execute the script? Could the problem be that `/bin/sh` is in a different location? The other problem could be that we didn't add `chmod a+x /tmp/script`, but I kinda doubt that.

powerjg, 2025-06-30:
In https://github.com/gem5/gem5/blob/develop/configs/example/gem5_library/riscv-ubuntu-run.py I don't see a `--script` argument. What changes to that file have you made?

I believe the follow change should work (after the `set_workload` call)

```python
board._set_readfile_contents("""
#!/bin/sh
echo "Hello from inside the simulated system!"
""")
```

Or

```python
board.readfile = "my_script.rcS"
```

Or, replace the `set_workload` call

```python
workload = obtain_resource("riscv-ubuntu-24.04-boot", resource_version="1.0.0")
workload.set_parameter("readfile", "my_script.rcS")
board.set_workload(workload)
```

Or, similarly

```python
workload = obtain_resource("riscv-ubuntu-24.04-boot", resource_version="1.0.0")
workload.set_parameter("readfile_contents", 
"""#!/bin/sh
echo "Hello from inside the simulated system!
""")
board.set_workload(workload)
```

I haven't tested any of these, but this is the *intention*. If one (or any) of these don't work, this is a bug that needs to be fixed.

powerjg, 2025-07-01:
Thanks for working with me on this. I just wanted to let you know that I've reproduced the problem. No ETA on a solution, though :).

Oddly, it seems that the readfile implementation in gem5 is getting valid data (e.g., the file path seems correct), but for some reason the correct data is not showing up in the `write` system call in the guest m5 binary.

ntampouratzis, 2025-07-06:
Dear @powerjg, the default gem5 Linux kernel version now is `riscv-linux-6.6.33-kernel`. I used this version in a previous gem5 version (24.1.x), and the `--script` works correctly.

In addition, I successfully built a new Linux kernel (6.15.4) using this tutorial (https://github.com/gem5/gem5-resources/tree/stable/src/riscv-fs/riscv-opensbi) with the latest gem5 version (v25), but the `--script` problem persists. 

Based on the above, I think the problem is not in the kernel. In the official gem5 release notes (https://github.com/gem5/gem5/blob/stable/RELEASE-NOTES.md), I read the following:

`The addition of the gem5 bridge driver means that sudo is no longer needed to run gem5-bridge commands (https://github.com/gem5/gem5/pull/1480)`

Is it possible that the problem is related to the new gem5 bridge driver?

saul44203, 2025-07-23:
Hey guys, If someone could verify #2464 fixes this it would be great. From my part it seems to work.

The readfile pseudoinst uses a functional translation and because the physical address was not being updated properly the virtual address was being used to proceed with the access.

It's a simple fix, I think this was left of in #1387.

Harshil2107, 2025-07-23:
I tested the #2464 PR with the `riscv-ubuntu-24.04-boot-no-systemd` V2.0.0 workload and I was able to run the test script via readfile. I am also running the `riscv-ubuntu-20.04-boot` V3.0.0 as a sanity check but it looks like PR #2464 fixes this issue.

commit 415cbec5 (2025-08-11): arch-riscv: populate logBytes/paddr after functional pt walk (#2464)

Both `logBytes` and `addr` references were not updated after finishing a
functional translation walk in the RISC-V page table walker.

Should fix #2410.