gem5/gem5 #2414: resources: GEM5_RESOURCE_JSON_APPEND doesn't work with gem5 v25.0

erin-le, 2025-07-01:
**Describe the bug**

On gem5 v25.0/ after the migration to Azure, running `export GEM5_RESOURCE_JSON_APPEND=/path/to/json/file` doesn't work. It looks like gem5 pings Azure for the workload instead of checking for it locally, then exits when it fails to find it on Azure.

**Affects version**
v25.0, stable branch

**gem5 Modifications**
I added a cache to the stdlib, but that should have no effect on this error.

**To Reproduce**
Steps to reproduce the behavior. Please assume starting from a clean repository:

1. Compile gem5 on the current `stable` branch
2. Compile the following C++ code into an X86 binary:
```cpp
#include <iostream>
#include <string>

int main(int argc, char* argv[]){
    std::cout << "Hello world! This is a custom resource to test the gem5 "
                "website's gem5-stdlib documentation, part 5." << std::endl;
    std::cout << "Now printing argv: " << std::endl;
    for (int i = 0; i < argc; i++) {
        std::cout << argv[i] << std::endl;
    }
    return 0;
}
```
3. Save the following JSON and modify the "url" field in "hello-world-testing-stdlib" to be the path of the binary from part 2:
```json
[
    {
        "category": "binary",
        "id": "hello-world-testing-stdlib",
        "description": "An X86 binary that prints hello world and an additional sentence",
        "architecture": "X86",
        "is_zipped": false,
        "resource_version": "1.0.0",
        "gem5_versions": [
            "25.0"
        ],
        "url": "file:///home/path/to/part/2/binary"
    },
    {
        "id": "hello-world-testing-stdlib-workload",
        "category": "workload",
        "description": "An X86 binary that prints hello world and an additional sentence",
        "architecture": "X86",
        "function": "set_se_binary_workload",
        "resource_version": "1.0.0",
        "gem5_versions": [
            "25.0"
        ],
        "resources": {
            "binary": {
                "id": "hello-world-testing-stdlib",
                "resource_version": "1.0.0"
            }
        },
        "additional_params": {
            "arguments": ["arg1", "arg2"]
        }
    }
]
```

4. Run `export GEM5_RESOURCE_JSON_APPEND=/path/to/part/3/json/file` in the terminal window you want to test this in.
5. Run the following gem5 config script with the `stable` build from part 1:
```python
from gem5.components.boards.simple_board import SimpleBoard
from gem5.components.cachehierarchies.classic.no_cache import NoCache
from gem5.components.memory.single_channel import SingleChannelDDR3_1600
from gem5.components.processors.cpu_types import CPUTypes
from gem5.components.processors.simple_processor import SimpleProcessor
from gem5.isas import ISA
from gem5.resources.resource import obtain_resource
from gem5.simulate.simulator import Simulator

# Obtain the components.
cache_hierarchy = NoCache()
memory = SingleChannelDDR3_1600("1GiB")
processor = SimpleProcessor(cpu_type=CPUTypes.ATOMIC, num_cores=1, isa=ISA.X86)

# Add them to the board.
board = SimpleBoard(
    clk_freq="3GHz",
    processor=processor,
    memory=memory,
    cache_hierarchy=cache_hierarchy,
)

# Set the workload.
binary = obtain_resource("hello-world-testing-stdlib-workload")
board.set_workload(binary)

# Setup the Simulator and run the simulation.
simulator = Simulator(board=board)
simulator.run()
```

**Terminal Output**
This is the output from `simerr.txt`

```txt
warn: Attempt 1 of Azure functions HTTP Request failed.
Purpose of Request: Get Resources.
Failed with Exception: HTTP Error 404: Not Found
Response: No requested resources were found

Retrying after 2 seconds...

warn: Attempt 2 of Azure functions HTTP Request failed.
Purpose of Request: Get Resources.
Failed with Exception: HTTP Error 404: Not Found
Response: No requested resources were found

Retrying after 4 seconds...

Exception thrown while getting resources '[<gem5.resources.client_api.client_query.ClientQuery object at 0x78f03dfdf2c0>]' from client 'gem5-resources'

AzureFunctionsAPIClientHttpJsonRequestError: Http Request to Azure Functions API failed.
Azure Functions API URL: https://api.gem5.org/api/resources
Data sent:

[
    {
        "id": "hello-world-testing-stdlib-workload",
        "resource_version": "None"
    }
]
Response: No requested resources were found
Purpose of Request: Get Resources
```

**Expected behavior**
For gem5 to be able to find the workload when using `GEM5_RESOURCE_JSON_APPEND`.

**Host Operating System**
Ubuntu 24.04

**Host ISA**
X86

**Additional information**
Exporting `GEM5_RESOURCE_JSON` still works, as does creating another JSON file and exporting `GEM5_CONFIG`.

zmf17, 2025-08-04:
感觉是顺序问题，如果设置 GEM5_RESOURCE_JSON 而不是 GEM5_RESOURCE_JSON_APPEND，可以正常工作

powerjg, 2025-09-17:
@Harshil2107 can you take a look?

closed 2025-11-18 by Harshil2107