gem5/gem5 #2454: RISCV: when performing PTW on an illegal address, it may cause the PTW to crash and be unable to proceed

yichenhit, 2025-07-17:
**Describe the bug**
In RISC-V ISA, when the MMU performs a new PTW that was previously blocked, it will process function _startWalkWrapper()_(src/arch/riscv/pagetable_walker.cc):
```void
Walker::startWalkWrapper()
{
    unsigned num_squashed = 0;
    WalkerState *currState = currStates.front();

    // check if we get a tlb hit to skip the walk
    Addr vaddr = Addr(sext<SV39_VADDR_BITS>(currState->req->getVaddr()));
    Addr vpn = getVPNFromVAddr(vaddr, currState->satp.mode);
    TlbEntry *e = tlb->lookup(vpn, currState->satp.asid, currState->mode,
                              true);
    Fault fault = NoFault;
    if (e) {
       fault = tlb->checkPermissions(currState->tc, currState->memaccess,
                            e->vaddr, currState->mode, e->pte);
    }

    while ((num_squashed < numSquashable) && currState &&
           (currState->translation->squashed() || (e && fault == NoFault))) {
        currStates.pop_front();
        num_squashed++;

        DPRINTF(PageTableWalker, "Squashing table walk for address %#x\n",
            currState->req->getVaddr());

        // finish the translation which will delete the translation object
        if (currState->translation->squashed()) {
            currState->translation->finish(
                std::make_shared<UnimpFault>("Squashed Inst"),
                currState->req, currState->tc, currState->mode);
        } else {
            tlb->translateTiming(currState->req, currState->tc,
                                 currState->translation, currState->mode);
        }

        // delete the current request if there are no inflight packets.
        // if there is something in flight, delete when the packets are
        // received and inflight is zero.
        if (currState->numInflight() == 0) {
            delete currState;
        } else {
            currState->squash();
        }

        // check the next translation request, if it exists
        if (currStates.size()) {
            currState = currStates.front();
            vaddr = Addr(sext<SV39_VADDR_BITS>(currState->req->getVaddr()));
            Addr vpn = getVPNFromVAddr(vaddr, currState->satp.mode);
            e = tlb->lookup(vpn, currState->satp.asid, currState->mode,
                            true);
            if (e) {
                fault = tlb->checkPermissions(currState->tc,
                    currState->memaccess, e->vaddr, currState->mode, e->pte);
            }
        } else {
            currState = NULL;
        }
    }
    if (currState && !currState->wasStarted()) {
        if (!e || fault != NoFault) {
            Fault timingFault = currState->walk();
            if (timingFault != NoFault) {     // Bug is here
                currStates.pop_front();
                delete currState;
                currState = NULL;
            }
        }
        else {
            schedule(startWalkWrapperEvent, clockEdge(Cycles(1)));
        }
    }
}```

In this function, when a _currState_ not find a tlb entry or _checkPermissions()_ return a fault, it will call _walk()_ and futher call _walkGStage()_, _walkTwoStage()_ or _walkOneStage()_. Among these three functions, the legitimacy of the virtual address will be checked first, just like:
```    // Make sure MSBS are the same
    // riscv-privileged-20211203 page 84
    auto mask_for_msbs = mask(64 - SV39_VADDR_BITS);
    auto msbs = bits(vaddr, 63, SV39_VADDR_BITS);
    if (msbs != 0 && msbs != mask_for_msbs) {
        return pageFault();
    }```

If return from here with a pagefault, MMU will not send any packet and _currState_ will be deleted. _startWalkWrapperEvent_ not be scheduled and MMU will never continue to next one PTW.

This may be a bug?

**Affects version**
Gem5 v25.0.0

**gem5 Modifications**
I discovered this bug while using a custom kernel plus workload. The logic of this part of PTW has not been modified. A typical problem scenario is: the program executes on the wrong path, resulting in an illegal virtual address

**Host Operating System**
Ubuntu 22.04

**Host ISA**
X86

**Additional information**
Add any other information which does not fit in the previous sections but may be of use in fixing this bug.

yichenhit, 2025-07-18:
> Did this work in gem5 v24.1? I.e., is this a regression or was it always a problem?

I checked the code of Tag v24.1.0.3-0-gb9da2bfe1e and there was no such problem.

powerjg, 2025-07-18:
Either one seems reasonable to me. I think that squashing all of them in a single cycle is probably reasonable. 

@yichenhit Do you want to create a PR?

yichenhit, 2025-07-22:
> Nice catch [@yichenhit](https://github.com/yichenhit) !  很敏锐！
> 
> This arises from the fact that before v25.0 (which includes [#1387](https://github.com/gem5/gem5/pull/1387) ) startWalk() which was renamed to walk() could never throw a fault.这是由于在 v25.0 版本之前（包括 [#1387](https://github.com/gem5/gem5/pull/1387) 版本），startWalk()函数被重命名为 walk()函数，该函数永远不会抛出异常。
> 
> The code snippet below (which you correctly identified) changes that. There is now a codepath that throws a pagefault before stepWalk().下面的代码片段（您已正确识别）改变了这一点。现在有一个代码路径会在 stepWalk()之前抛出页错误。
> 
> // Make sure MSBS are the same
> // riscv-privileged-20211203 page 84
> auto mask_for_msbs = mask(64 - SV39_VADDR_BITS);
> auto msbs = bits(vaddr, 63, SV39_VADDR_BITS);
> if (msbs != 0 && msbs != mask_for_msbs) {
>     return pageFault();
> }
> Our handling of this case is erroneous as you identified because the next walks that the PTW has to perform are not started, we simply discard the faulty one.正如您所识别的，我们对此情况的处理是错误的，因为 PTW 接下来需要执行的遍历并未开始，我们只是简单地丢弃了有问题的遍历。
> 
> I think your fix addresses the problem, however I am not sure if the 1 cycle penalty for each faulty walk is realistic. Perhaps the PTW can "pop" all faulty walks at once and start the next walk.我认为您的修复方案解决了问题，但我不确定每次有问题的遍历的 1 周期惩罚是否现实。或许 PTW 可以一次性“弹出”所有有问题的遍历，并开始下一次遍历。
> 
> Something like the following:类似以下内容：
> 
> if (currState && !currState->wasStarted()) {
>     if (!e || fault != NoFault) {
>         Fault timingFault = currState->walk();
>     
>         // Catch all walker states that have early fault!
>         while (!currStates.empty() && timingFault != NoFault) {
>             // Delete currState due to early fault
>             currStates.pop_front();
>             delete currState;
>     
>             // Get next currState & walk
>             if (currStates.size() > 0) {
>                 currState = currStates.front();
>                 timingFault = currState->walk();
>             }
>         }
>     }
>     else {
>         schedule(startWalkWrapperEvent, clockEdge(Cycles(1)));
>     }
> }
> [@powerjg](https://github.com/powerjg) I am not certain which fix would be most realistic, maybe you could comment on that.我不确定哪种修复方案最现实，或许你可以对此发表评论。

This solution would be more reasonable. My solution merely made simple modifications to this bug

yichenhit, 2025-07-22:
> Either one seems reasonable to me. I think that squashing all of them in a single cycle is probably reasonable.在我看来，这两种方案似乎都合理。我认为将它们全部在一个周期内解决可能是合理的。
> 
> [@yichenhit](https://github.com/yichenhit) Do you want to create a PR?你想创建一个 PR 吗？

Okay，I never contribute to gem5 before, I could try this time

closed 2025-08-06 by yichenhit