gem5/gem5 #2461: Exit handlers with m5ops not working properly

alisemi, 2025-07-21:
**Describe the bug**
I am simulating a program that uses m5opt to mark the region of interest (ROI) with m5_work_begin and m5_work_end calls. I tried to incorporate the new exit handlers to define the behaviour  but they don't work as expected, instead I am getting the usual warning behaviours:
warn: No behavior was set by the user for work begin. Default behavior is resetting the stats and continuing.
warn: No behavior was set by the user for work end. Default behavior is dumping the stats and continuing.

```python
# class MySchedHandler(WorkBeginExitHandler):
#     def _process(self, simulator):
#         super()._process(simulator)
#         print("Got this scheduled event")
#         print(f"the current tick is {simulator.get_current_tick()}")
#         print(f"justification is {self.justification()}")

#     def _exit_simulation(self):
#         return False

def do_a_thing(simulator, payload):
    print("Doing a thing")
    print(f"payload is {payload}")
    print(f"the current tick is {simulator.get_current_tick()}")
    return False

register_exit_handler(4, do_a_thing, description="Do a thing handler")

sim = Simulator(
    board=board
)

sim.run()
```

I tried both using the class defining method ( e.g. ```class MySchedHandler(WorkBeginExitHandler):``` ) and ```register_exit_handler```, but none worked. Am I missing something in the configuration maybe?

**Affects version**
v25.0.0.0 stable branch.

**gem5 Modifications**
No modifications

**To Reproduce**
1. Compile gem5.opt and m5 on X86 architecture, i.e.  `scons build/X86/gem5.opt -j8; scons build/x86/out/m5 -j8;`  
2. Adjust the Makefile in the attached example program for GEM5_HOME directory, compile it using `make` command.
3. Change the binary_path in the attached exit_handler_issue.py configuration to point to the example program.
4. cd into gem5 directory.
5. Using the attached python configuration script, run the simulation with:
```
shell ./build/X86/gem5.opt  -re ./configs/bug/exit_handler_issue.py --num_cores 4
```

**Expected behavior**
Custom behaviours as registered by the custom callbacks.

**Host Operating System**
CentOS Linux 7 (Core)

**Host ISA**
X86

**Compiler used**
gcc version 10.3.0 (GCC) 
scons v3.1.0

**Additional information**
I am attaching the full configuration and program files for testing.

[exit_handler_issue.zip](https://github.com/user-attachments/files/21345035/exit_handler_issue.zip)

alisemi, 2025-07-22:
Thanks @Harshil2107! It works with m5_hypercall functions work well. I am closing the issue then.

closed 2025-07-22 by alisemi

powerjg, 2025-07-22:
@Harshil2107, did we never push the enum/defines for these hypercalls? E.g., can we not do something like `m5_hypercall(WORK_BEGIN);`?

Harshil2107, 2025-07-23:
I have created a PR that adds a header file for the hypercalls #2467