gem5/gem5 #2475: arch-arm: attempted to execute unknown instruction

yichungchen84, 2025-07-24:
**Describe the bug**
panic: attempted to execute unknown instruction  

after objdump
"urhadd z0.b, p0/m, z0.b, z1.b" <= this instruction 

**Affects version**
v25.0.0.0

**gem5 Modifications**
none

**To Reproduce**
use SE mode for running checkpoint creation
SPECint2017 525

**Host Operating System**
ubuntu 24.04

**Host ISA**
X86_64

**Compiler used**
LLVM16

yichungchen84, 2025-07-24:
> Could you report the encoding of the instruction? It should be printed alongside the panic message

@giactra I add more information for your reference. It should print out "arm_inst->encoding()", but there is nothing. 

Meanwhile I will try to see if ArmNonCachingSimpleCPU works. Or try 23.1.0.0

giactra, 2025-07-25:
Hi @yichungchen84 , those are SVE2 instructions which we currently don't support in gem5.
Your workload might think the CPU implements FEAT_SVE2 as it supports FEAT_SME in SE mode (as you can see here https://github.com/gem5/gem5/blob/stable/src/arch/arm/ArmISA.py#L78), and SME requires SVE2 to be present.

With respect to it, gem5 is not really architecturally compliant.

For the time being (until we implement SVE2), I suggest you to disable FEAT_SME in by providing a different arch release to the 
[release_se parameter](https://github.com/gem5/gem5/blob/stable/src/arch/arm/ArmISA.py#L96C5-L96C15)

Let me know if this fixes your problem

yichungchen84, 2025-07-29:
Hi @giactra, I tried to disable the FEAT_SME. Unfortunately, it has the same panic. Do you happen to have other suggestion? Thanks.

giactra, 2025-07-29:
Did you compile the SPEC2017 simpoint yourself? If yes would it be possible to check the compile flags and remove any sve2/sme instance?

Alternatively, we have an internal WIP implementation of SVE2 we plan to upstream in the following months

yichungchen84, 2025-07-30:
Hi, @giactra. 
Unfortunately, I did not generate binaries myself and I know there is sve2 cflags. The binaries were generated for understanding performance limitation of Arm X9xx cores.

Thanks for letting me know you are working on SVE2 model. I look forward to it. Meanwhile, if I can be of any help, please feel free to let me know.

(I will leave this issue open and come back later.)