gem5/gem5 #2530: gem5 O3 CPU an assertion failed Bug Report

30ZHANGjin, 2025-08-14:
**Describe the bug**
The simulation aborts with an assertion failure in src/cpu/o3/dyn_inst.cc. The assertion that fails is cpu->instCount() == 0 (or cpu->instCount() <= 1500 in other versions), indicating that an instruction counter was not properly reset before initializing a new dynamic instruction, likely after a pipeline squash.

This issue appears to be a latent bug in the O3 core's squash logic that is exposed under specific conditions. It is not triggered with a baseline 8-wide configuration, but consistently occurs when the core's frontend/backend width is increased significantly (e.g., to 12) and internal resources (ROB, IQ, FUs) are scaled up accordingly. 
**Affects version**
gem5-24-000

**gem5 Modifications**
No modifications were made to the gem5 source code. The issue was reproduced using standard gem5 configuration scripts (configs/example/se.py) with modified parameters passed via the command line or a custom configuration script to create a wider, deeper O3 CPU model.

**Terminal Output**

```shell
Program aborted at tick 8568017655000
--- BEGIN LICK BACKTRACE ---
/home/zhangwenjin/gem5/my-simpoint-spec06-gem5-24/build/RISCV/gem5.opt(_Z15print_backtracev+0x3d) [0x40d3a153d8]
/home/zhangwenjin/gem5/my-simpoint-spec06-gem5-24/build/RISCV/gem5.opt(_Z22AbortHandlerExt(int, P9__siginfo, Pv)+0x4c) [0x40d4a1a7056c]
/lib/x86_64-linux-gnu/libc.so.6(+0x453a0) [0x7d6b176c63a0]
...
gem5.opt: src/cpu/o3/dyn_inst.cc:83: void gem5::o3::DynInst::init(...): Assertion 'cpu->instCount() == 0' failed.
...
(gdb) backtrace
#0  0x0000... in raise () from /lib/x86_64-linux-gnu/libc.so.6
#1  0x0000... in abort () from /lib/x86_64-linux-gnu/libc.so.6
#2  0x0000... in gem5::o3::DynInst::init(...) at src/cpu/o3/dyn_inst.cc:83
#3  0x0000... in gem5::o3::Fetch::buildInst(...) at src/cpu/o3/fetch.cc:1085
#4  0x0000... in gem5::o3::Fetch::tick() at src/cpu/o3/fetch.cc:1893
#5  0x0000... in gem5::o3::CPU::tick() at src/cpu/o3/cpu.cc:375
#6  0x0000... in gem5::Clocked::ClockedEvent::process() at src/sim/clocked.cc:152
#7  0x0000... in gem5::Event::process() at src/sim/eventq.cc:240
#8  0x0000... in gem5::EventQueue::serviceOne() at src/sim/eventq.cc:1162
#9  0x0000... in gem5::simulate(long) at src/sim/simulate.cc:234
...

```

**Expected behavior**
The simulation should either complete successfully or, if an unrecoverable state is reached, fail with a more descriptive panic message related to the specific microarchitectural inconsistency, rather than this low-level assertion failure. The current assertion suggests a state management issue within the squash logic itself.

**Host Operating System**
Ubuntu 24.04，windows

**Host ISA**
X86

**Compiler used**
gcc 10.5

**Additional information**
he key finding is that this bug seems to be a scaling issue. It is not present in baseline, narrower O3 core configurations. It only manifests when the number of in-flight instructions that need to be handled during a squash event becomes very large, due to the increased width and resource depth of the core. This points towards a potential flaw in how the squash mechanism handles state reset under high pressure.

yichungchen84, 2025-08-17:
@30ZHANGjin this is not a bug. You can change it to a number >1500. The number should be as reasonable as possible since it could help you to debug modeling logic errors. If you are working on modeling of large modern CPUs. This and FU limit (cpu/o3/limit.h) are two fixes after downloading a new GEM5.

30ZHANGjin, 2025-09-18:
> I have this problem with the latest version (v25.0.0.1), in O3 core. I have not changed any cpu configuration, and I use the baseline SimpleProcessor for RISC-V. The only difference to a base system is that I use my own custom kernel and disk image, where I have enabled systemd in order to bringup docker. The `cpu->instCount() <= 1500` assertion is triggered during systemd initialization phase after kernel boot. After changing the limit to 5000 insts, the simulated system finishes boot successfully. So there might actually be a bug with decrementing the counter. (I don't know if it is a possibility for you to build a disk image with systemd to reproduce the problem.)

I didn't try to reproduce the problem again because increasing the cpu->instCount() value resolved the error for me. thank you