gem5/gem5 #2553: Unexpected tmp_d0 appears after replacing Vs3_ub in vse8_v with Vs3.

buloruabutata, 2025-08-26:
I am trying to template the store instruction in the V extension, and the following is my modification.
When I replace Vs3_ub with VS3, tmp_d0 will appear in %(op_rd)s.
In src/arch/isa_parser/operand_types.py, the `is_dest` of Vs3 is set to True, which causes the `makeReadW` method to be called.
An alternative is to set is_dest to false if Vs3 is detected when VecRegOperand is initialized. But I wonder if there is any other better way.
```
0x00: VseOp::vse8_v({{
                            auto Vs3 = tmp_s1.as<uint8_t>();
                            Mem_vc.as<uint8_t>()[i] = Vs3[i];
                        }}, inst_flags=SimdUnitStrideStoreOp);
```

erin-le, 2025-08-29:
@saul44203: Hello, could you take a look at this issue? Thank you!

buloruabutata, 2025-08-29:
Maybe adding et to operand_types in src/ARCH/RISCV/ISA/Operands. ISA or adding vu to the template can solve the problem. I'm trying.
```
def operand_types {{
    'vi'    : 'vi',
    'vu'    : 'vu',
    'et'    : 'ElemType',
}};
```
```
0x00: VseOp::vse8_v({{
                            Mem_vc.as<vu>()[i] = Vs3_vu[i];
                        }}, inst_flags=SimdUnitStrideStoreOp);
```
```
0x00: VseOp::vse8_v({{
                            Mem_vc.as<et>()[i] = Vs3_et[i];
                        }}, inst_flags=SimdUnitStrideStoreOp);
```